flightaware/piaware #12: piaware dies if faup1090 is missing

mutability, 2015-05-27:
Presumably this should be caught before it propagates all the way out and kills the process.

````
pi@twopi ~/piaware $ sudo piaware -debug
05/27/2015 19:02:39 ****************************************************
05/27/2015 19:02:39 piaware version 1.20 is running, process ID 10541
05/27/2015 19:02:39 your system info is: Linux twopi 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv6l GNU/Linux
05/27/2015 19:02:39 connecting to FlightAware eyes.flightaware.com/1200
05/27/2015 19:02:40 FlightAware server SSL certificate validated
05/27/2015 19:02:40 encrypted session established with FlightAware
05/27/2015 19:02:41 autoUpdate is not configured in /etc/piaware or by piaware-config
05/27/2015 19:02:41 manualUpdate is not configured in /etc/piaware or by piaware-config
05/27/2015 19:02:41 ADS-B data program 'dump1090-mutab' is listening on port 30005, so far so good
05/27/2015 19:02:41 i see nothing serving on port 10001, starting faup1090...
couldn't execute "/usr/bin/faup1090": no such file or directory
    while executing
"exec /usr/bin/faup1090 &"
    (procedure "start_faup1090" line 2)
    invoked from within
"start_faup1090"
    (procedure "connect_fa_style_adsb_port" line 19)
    invoked from within
"connect_fa_style_adsb_port"
    (procedure "setup_fa_style_adsb_client" line 6)
    invoked from within
"setup_fa_style_adsb_client"
    (procedure "main" line 86)
    invoked from within
"main $argv"
    invoked from within
"if {!$tcl_interactive} {
    main $argv
}"
    (file "/usr/lib/piaware/main.tcl" line 127)
    invoked from within
"source /usr/lib/piaware/main.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel #0 source $path"
    from tcllauncher running "piaware -debug"
pi@twopi ~/piaware $
````

anish, 2015-06-11:
Isn't faup1090 gone with 2.0?

dbaker, 2015-06-11:
obj: I agree.  Is your thinking that if there's no fadump1090 installed, the program should just keep retrying 10001?

anish: faup1090 is still required for 2.0 if you're running a non-FlightAware dump1090.

mutability, 2015-06-11:
Right, it obviously isn't going to work right until it's fixed, but piaware should at least keep running and retrying.

anish, 2015-06-11:
@dbaker not sure I get you. The dump1090_mr repo that used to have faup1090 is gone now, did the source move elsewhere ?

mutability, 2015-09-01:
This is no longer a problem with 2.1 as the faup1090-starting code all changed anyway.

For the record the old faup1090 for 2.0 is at https://github.com/flightaware/dump1090/tree/legacy_dump1090_mr and the new faup1090 for 2.1 is built from dump1090-mutability at https://github.com/mutability/dump1090

closed 2015-09-01 by mutability