flightaware/piaware #20: Cosmetic issues: Absoulte path to helper programs displayed incorrectly.

dschaper, 2015-07-22:
Cosmetic fix only. 
Running ps ax displays /usr/bin/python3 /usr/sbin/fa-mlat-client but the client is actually located in /usr/bin

locate fa-mlat-client
/usr/bin/fa-mlat-client

Also, logging displays `/usr/sbin` for faup1090 and fa-mlat-client.

mutability, 2015-07-22:
What piaware version? The current package installs nothing in /usr/sbin and installs fa-mlat-client under /usr/lib/fa-mlat-client. If you have stuff in /usr/bin then it's not from a current piaware

mutability, 2015-07-22:
FWIW, a sdcard image piaware install yields this as expected:

```
$ ps ax | grep fa-mlat-client
 3988 ?        S      8:42 /usr/lib/fa-mlat-client/fa-mlat-client [... other args ...]
```

dschaper, 2015-07-22:
piaware -v
    2.1-2

    ps ax | grep fa-mlat-client
    31429 ?        S      0:51 /usr/bin/python3 /usr/sbin/fa-mlat-client

    piaware version 2.1-1 is running, process ID 16853
    Starting faup1090: /usr/sbin/faup1090 
    Starting multilateration client: /usr/sbin/fa-mlat-client

I'll try a re-install. Everything was from pulled from git for the latest clients. Maybe I compiled with a prefix somewhere that threw everything off. Also running on Arch, but I don't know how that could be a cause?

dschaper, 2015-07-22:
I'll grab a copy of the image and see why there are diffs. Probably a user error, will try to git and compile to match the image paths. There may be issues with the PKGBUILD's and not with PiAware.

dschaper, 2015-07-22:
Sorry to add noise to this but would a raspbian-ua-netinstall followed by a piaware_builder give me the same setup as going with the image?

dschaper, 2015-07-23:
This is kind of strange, if piaware is started from the command line, the paths are okay. If piaware is started from sytemctl using the supplied service file then the paths show /usr/sbin even though there are no programs in /usr/sbin?

mutability, 2015-07-23:
Oh, ok. systemd. I don't really know anything about systemd.

piaware tries to find fa-mlat-client at /usr/lib/fa-mlat-client/fa-mlat-client. If that doesn't exist it will look in $PATH. It will also look in $PATH for faup1090. So I can only guess that something that systemd does ends up making them available in /usr/sbin (maybe it builds a little chroot or something? I have no idea..)

dschaper, 2015-07-23:
I'll research that and see what I can find.

dschaper, 2015-08-07:
Recent update to new paths for helper programs and removal of path search fixed issue.

closed 2015-08-07 by dschaper

mutability, 2015-08-07:
Weird. Dunno how that changed anything.

dschaper, 2015-08-07:
Had to change the PKGBUILD's (Arch). Now the helper applications are in the right directories instead of being left in /usr/bin. (Forced the applications to be installed correctly, or else piaware wouldn't function.)