flightaware/piaware #26: Okay to use dev versions?

dschaper, 2016-01-20:
I finally have gpsd up and running on the Orange Pi. Is it okay to run the dev version, or would you prefer that not be run until further testing is done? I'm aware that it may break my setup, I just don't want to break FA's setup. I'd be happy to beta the software for you as another data point.

mutability, 2016-01-20:
If you're going to try it I would use a site that you don't care about; some of the changes are likely to break the site on the FA side in ways that you currently can't repair yourself (e.g. currently if you ever feed a receiver position, then you can no longer manually set the site location in the web interface)

dschaper, 2016-01-20:
I don't mind if my side breaks, I'm not a stats chaser really. I just don't want to cause any trouble on the FA side of things.

mutability, 2016-01-20:
Well the dev version identifies itself as a dev version, so FA can throw away that data if needed (it's actually something I've been meaning to do..)

dschaper, 2016-01-20:
Okay, thanks! I'll give it a try later and see how it goes.

dschaper, 2016-01-20:
Built 9a5f8aefcd5924595fa73c0a1d4f78d002ef3eed via piaware-builder. (Needed to add a dummy entry to changelog for the 2.2.0~dev.) Log dump follows:
```
orangepi@OrangePI:~$ cat /tmp/piaware.out
01/20/2016 18:58:14 ****************************************************
01/20/2016 18:58:14 piaware version 2.2.0~dev is running, process ID 956
01/20/2016 18:58:14 your system info is: Linux OrangePI 3.4.39-01-lobo #1 SMP PREEMPT Sun Oct 25 14:46:41 CET 2015 armv7l armv7l armv7l GNU/Linux
01/20/2016 18:58:14 GPS: Connected to gpsd 3.11 at localhost/2947
01/20/2016 18:58:15 Connecting to FlightAware adept server at piaware.flightaware.com/1200
01/20/2016 18:58:16 Connection with adept server at piaware.flightaware.com/1200 established
01/20/2016 18:58:16 FlightAware server SSL certificate validated
01/20/2016 18:58:16 encrypted session established with FlightAware
01/20/2016 18:58:16 autoUpdate is not configured in /etc/piaware or by piaware-config
01/20/2016 18:58:16 manualUpdate is not configured in /etc/piaware or by piaware-config
01/20/2016 18:58:16 multilateration support enabled (use piaware-config to disable)
01/20/2016 18:58:16 Receiver location changed, restarting dump1090
01/20/2016 18:58:16 attempting to restart dump1090..
01/20/2016 18:58:17 error handling message 'clock\t1453316296\trecv_lat\t33.798421\trecv_lon\t-117.798358\tstatus\tok\ttype\tlogin_response\tuser\tdschaper' from server (can't read "services": no such variable), (can't read "services": no such variable\n    while executing\n"llength $services"\n    (procedure "attempt_service_restart" line 18)\n    invoked from within\n"attempt_service_restart dump1090 restart"\n    (procedure "update_location" line 29)\n    invoked from within\n"update_location $lat $lon"\n    (procedure "adept_location_changed" line 8)\n    invoked from within\n"adept_location_changed $row(recv_lat) $row(recv_lon)"\n    (object "::adept" method "::fa_adept::AdeptClient::handle_login_response_message" body line 20)\n    invoked from within\n"handle_login_response_message row"\n    ("login_response" arm line 2)\n    invoked from within\n"switch -glob $row(type) {\n\t\t\t"login_response" {\n\t\t\t\thandle_login_response_message row\n\t\t\t}\n\n\t\t\t"notice" {\n\t\t\t\thandle_notice_message row\n\t\t\t}\n\n\t\t\t"aliv..."\n    (object "::adept" method "::fa_adept::AdeptClient::handle_response" body line 4)\n    invoked from within\n"handle_response response"), disconnecting and reconnecting...
01/20/2016 18:58:17 TLS alert: close notify
01/20/2016 18:58:17 reconnecting in 116 seconds...
01/20/2016 18:58:17 ADS-B data program 'dump1090-mutabi' is listening on port 30005, so far so good
01/20/2016 18:58:17 Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30005 --stdout --lat 33.798 --lon -117.798
01/20/2016 18:58:17 Started faup1090 (pid 997) to connect to dump1090-mutabi
01/20/2016 18:58:17 GPS: Receiver location: 33.79840, -117.79844 at 112m height (WGS84)
01/20/2016 18:58:18 piaware received a message from dump1090-mutabi!
01/20/2016 18:58:49 43 msgs recv'd from dump1090-mutabi; 0 msgs sent to FlightAware
```

mutability, 2016-01-26:
That was fixed in f1145c954836a7223f4e8a535fbcae2c0c56bd4f

closed 2016-01-26 by mutability