flightaware/piaware #29: fa-mlat-client 0.2.4 crashes on local airspy_adsb (new version with beast this time)

jester-xbmc, 2016-02-17:
Opened a new issue as there is a new version by @touil with beast support (-f beast) here http://airspy.com/downloads/airspy_adsb-linux-arm-beast.tgz
The stream is pushing 20MHz MLAT, could that give the issues below ?

mlat(7442): Problem reading receiver messages: Out of range timestamp seen (last 4142366797, now 553222221)
02/17/2016 22:07:42 mlat(7442): Traceback (most recent call last):
02/17/2016 22:07:42 mlat(7442):   File "/usr/lib/python3.2/asyncore.py", line 83, in read
02/17/2016 22:07:42 mlat(7442):   File "/usr/lib/python3.2/asyncore.py", line 449, in handle_read_event
02/17/2016 22:07:42 mlat(7442):   File "/tmp/buildd/piaware-2.1/debian/venv/lib/python3.2/site-packages/mlat/client/receiver.py", line 129, in handle_read
02/17/2016 22:07:42 mlat(7442): ValueError: Lost sync with input stream: expected 0x1A after 0x1A escape

maybe issue like : mutability/mlat-client#1

jester-xbmc, 2016-02-17:
cross posting from the old closed ticket:

@mutability says:
I think it is generating the timestamps as little-endian; they should be big-endian.

ping  @touil

mutability, 2016-02-17:
The traceback indicates there might be a problem with the stream framing, too. I'd need to see the raw bytes to be sure.

20MHz means it won't work out-of-the-box with mlat-server. mlat-client should be mostly OK with it, but mlat-server will just never manage to get any sync with other receivers because the timing will be way off. I can add a new receiver type to mlat-client if it's always going to be 20MHz, but a simpler approach might be to make beast-format actually look like a Beast i.e. produce a 12MHz timestamp.

jester-xbmc, 2016-02-17:
new version fixes the endian issue (http://airspy.com/?ddownload=3753) but still this error:

02/17/2016 22:43:41 mlat(8338): Unexpected exception on connection to localhost:30005
02/17/2016 22:43:41 mlat(8338): Traceback (most recent call last):
02/17/2016 22:43:41 mlat(8338):   File "/usr/lib/python3.2/asyncore.py", line 83, in read
02/17/2016 22:43:41 mlat(8338):   File "/usr/lib/python3.2/asyncore.py", line 449, in handle_read_event
02/17/2016 22:43:41 mlat(8338):   File "/tmp/buildd/piaware-2.1/debian/venv/lib/python3.2/site-packages/mlat/client/receiver.py", line 129, in handle_read
02/17/2016 22:43:41 mlat(8338): ValueError: Lost sync with input stream: expected 0x1A after 0x1A escape

indeed might help to either default 12 MHZ for beast streams or add parameter to select this (in airspy_adsb)

touil, 2016-02-17:
May be I'm not sending something? what's the <esc><esc> sequence for?

mutability, 2016-02-17:
The Beast format is originally a serial data format where you might start mid-packet, so it does the equivalent of DLE-stuffing with 1A characters to make it easier to scan for a packet start. See http://wiki.modesbeast.com/Mode-S_Beast:Data_Output_Formats. If you need to send a 1A byte within the body of the message, then it should be sent as 1A 1A.

touil, 2016-02-17:
Does it apply to the timestamp as well?

mutability, 2016-02-17:
Yes, it applies to everything. The only un-doubled 1A is the very first byte.

touil, 2016-02-17:
OK. Got it and fixing. Thanks.

touil, 2016-02-17:
Looks fine now.

jester-xbmc, 2016-02-18:
looks like we have a winner! i'll keep it running overnight, no errors so far.
thanks !

touil, 2016-02-18:
Now that the Beast format is implemented, we will have a firmware update that allows 12MHz MLAT. Stay tuned!

mutability, 2016-02-18:
It would be fine to run at whatever rate you want and rescale that clock to 12MHz just for the Beast output. mlat-server just ends up scaling it by the clock frequency anyway. ADS-B based synchronization between receivers only gives about 0.5-1us accuracy so the rounding errors from rescaling aren't a big deal.

touil, 2016-02-18:
FYI. There will be another (cheaper) Airspy Mini device with can run at 12MSPS directly, so rescaling might be a good option to keep things simple.
In the other hand, I can add the re-scaling into the decoder itself and still get the nominal resolution at +/- 1 sample so there will be no degradation.
Home work for tonight!

jester-xbmc, 2016-02-18:
After running the new version for a few hours, its pretty stable, however it seems the mlat-client doesn't sync with other nearby servers (and yes there are several nearby)

02/18/2016 12:26:35 mlat(712): Receiver status: connected
02/18/2016 12:26:35 mlat(712): Server status:   not synchronized with any nearby receivers
02/18/2016 12:26:35 mlat(712): Receiver:  803.3 msg/s received     15.7kB/s from receiver
02/18/2016 12:26:35 mlat(712): Server:      0.0 kB/s from server    0.0kB/s TCP to server   2.1kB/s UDP to server
02/18/2016 12:26:35 mlat(712): Aircraft: 17 of 32 Mode S, 33 of 71 ADS-B used

touil, 2016-02-18:
That's expected. The MLAT is almost twice as fast in Airspy. Watch out for tonight's version.

touil, 2016-02-18:
Everything is implemented and working as expected.

jester-xbmc, 2016-02-18:
Getting closer but:

02/18/2016 22:31:20 mlat(5320): Receiver status: connected
02/18/2016 22:31:20 mlat(5320): Server status:   clock unstable
02/18/2016 22:31:20 mlat(5320): Receiver:  383.4 msg/s received      7.5kB/s from receiver
02/18/2016 22:31:20 mlat(5320): Server:      0.0 kB/s from server    0.0kB/s TCP to server   0.8kB/s UDP to server
02/18/2016 22:31:20 mlat(5320): Aircraft: 2 of 9 Mode S, 19 of 31 ADS-B used

mutability, 2016-02-18:
Yeah, definitely something wrong with the timing here.

The server takes pairs of even/odd DF17 position messages and pairs them up based on message contents + the interval between the messages. The message interval has to be within 1ms of the interval seen by other receivers, and they are sufficiently close that it's getting past that check. However after correcting for propagation delay from the aircraft to each receiver, the intervals disagree by a lot - 200ppm or more, and they jump around a lot.

touil, 2016-02-18:
Are you sure you are using v1.3?

jester-xbmc, 2016-02-18:
Yup, just downloaded:
airspy_adsb v1.3

touil, 2016-02-19:
May be it's the scaling code. Basically ts_12MHz = (ts_20MHz * 12 + 20 / 2) / 20. It should be fine within 1 sample

mutability, 2016-02-19:
I can't see anything obviously wrong with that scaling. Are you confident of the underlying timestamp?

mutability, 2016-02-19:
I have turned on a little more debugging on the server side so I can see the actual differences in intervals, but the client reconnected at around 0130 UTC and seems to have changed behaviour at that point, now it's back to not even passing the 1ms test.

touil, 2016-02-19:
I updated the RPi build if someone want to give it a try.

jester-xbmc, 2016-02-19:
uhm, now it goes back to the "old" message:

02/19/2016 14:14:57 mlat(19379): Receiver status: connected
02/19/2016 14:14:57 mlat(19379): Server status:   not synchronized with any nearby receivers
02/19/2016 14:14:57 mlat(19379): Receiver:  659.8 msg/s received     12.9kB/s from receiver
02/19/2016 14:14:57 mlat(19379): Server:      0.0 kB/s from server    0.0kB/s TCP to server   1.6kB/s UDP to server

touil, 2016-02-19:
@mutability Can you check the server side?

mutability, 2016-02-19:
@jester-xbmc: This is for mac address b8:27:eb:50:0a:97 right?

@touil:

```
20160219 14:40:23.191 master INFO     clocktrack           interval out of range, 997434.8333us vs 1449631.0000us
20160219 14:40:23.193 master INFO     clocktrack           interval out of range, 685604.7500us vs 1010058.7500us
20160219 14:40:23.194 master INFO     clocktrack           interval out of range, 1460863.9167us vs 2104718.2500us
20160219 14:40:23.282 master INFO     clocktrack           interval out of range, 1048143.4167us vs 1520012.0006us
20160219 14:40:23.289 master INFO     clocktrack           interval out of range, 726760.4167us vs 1070821.6667us
20160219 14:40:23.312 master INFO     clocktrack           interval out of range, 726760.4167us vs 1070824.7500us
20160219 14:40:23.330 master INFO     clocktrack           interval out of range, 1048143.4167us vs 1520040.5000us
20160219 14:40:23.339 master INFO     clocktrack           interval out of range, 695248.8333us vs 1056890.3333us
20160219 14:40:23.517 master INFO     clocktrack           interval out of range, 352520.5000us vs 500019.3333us
20160219 14:40:23.584 master INFO     clocktrack           interval out of range, 352520.5000us vs 499971.5001us
20160219 14:40:23.673 master INFO     clocktrack           interval out of range, 352520.5000us vs 499998.9167us
20160219 14:40:23.692 master INFO     clocktrack           interval out of range, 312953.6667us vs 450621.0834us
20160219 14:40:23.693 master INFO     clocktrack           interval out of range, 1999907.2500us vs 2869857.5000us
20160219 14:40:23.695 master INFO     clocktrack           interval out of range, 372910.7500us vs 540043.5000us
20160219 14:40:23.696 master INFO     clocktrack           interval out of range, 1843207.2500us vs 2659225.5001us
```

Lefthand numbers are intervals as measured by the airspy, righthand are intervals as measured by other receivers that it's trying to match to. There's no particularly consistent factor there I can see. Note that some of them may be real mismatches (trying to match up transmission times that aren't for the same messages, this can happen when an identical position message is sent at different times)

jester-xbmc, 2016-02-19:
@mutability correct.

jester-xbmc, 2016-02-19:
when running with /airspy_adsb -l 30005 -f beast -b -g 21 -p -d 2 -w 4
i'm getting MLAT data packets...

touil, 2016-02-19:
this means there were packet drops

mutability, 2016-02-19:
Whatever you changed when you reconnected at 1515 brought the inter-message intervals back into range and the frequency offset numbers look sensible. Now the server reckons your clock is unstable because it sees periodic large jumps in the absolute clock offset:

```
20160219 15:16:04.897 master INFO     clocksync            fAT-b827eb280072_000084:fAT-b827eb500a97_000162: fAT-b827eb500a97_000162 clock was 142547.7us slower than predicted, drift error -0.8ppm
```

which if you were running dump1090 would mean that you were losing large chunks of samples before they reached dump1090 itself. (dump1090 will jump the timestamp forward if it drops a block of samples due to overload, but in extreme overload cases the USB subsystem can start dropping data too). I guess something similar can happen with the airspy.

touil, 2016-02-19:
The threading and decimation options can help a lot here. Can you check with 4 workers?

jester-xbmc, 2016-02-19:
trying various -w and -d values, however after a number of MLAT messages it stills, clock drift ? packet loss ?

touil, 2016-02-19:
Packets are being dropped.

touil, 2016-02-19:
I'll optimize libairspy for the RPi. More efficient unpacking to come. Stay tuned!

touil, 2016-02-20:
@mutability Anything worth mentioning in the server side?

jester-xbmc, 2016-02-20:
piping airspy_adsb output through modesmixer gives unstable results, 15 minutes of clock unstable and then 15 minutes of "normal" running

02/20/2016 16:00:50 mlat(698): Receiver status: connected
02/20/2016 16:00:50 mlat(698): Server status:   synchronized with 126 nearby receivers
02/20/2016 16:00:50 mlat(698): Receiver:  682.2 msg/s received     12.7kB/s from receiver
02/20/2016 16:00:50 mlat(698): Server:      0.7 kB/s from server    0.0kB/s TCP to server   2.4kB/s UDP to server
02/20/2016 16:00:50 mlat(698): Results:  331.0 positions/minute
02/20/2016 16:00:50 mlat(698): Aircraft: 14 of 28 Mode S, 25 of 55 ADS-B used
02/20/2016 16:04:32 33242 msgs recv'd from modesmixer2 (1228 in last 5m); 32602 msgs sent to FlightAware
02/20/2016 16:09:32 34592 msgs recv'd from modesmixer2 (1350 in last 5m); 33952 msgs sent to FlightAware
02/20/2016 16:14:32 35992 msgs recv'd from modesmixer2 (1400 in last 5m); 35352 msgs sent to FlightAware
02/20/2016 16:15:51 mlat(698): Receiver status: connected
02/20/2016 16:15:51 mlat(698): Server status:   clock unstable
02/20/2016 16:15:51 mlat(698): Receiver:  736.3 msg/s received     13.7kB/s from receiver
02/20/2016 16:15:51 mlat(698): Server:      0.7 kB/s from server    0.0kB/s TCP to server   2.1kB/s UDP to server
02/20/2016 16:15:51 mlat(698): Results:  331.7 positions/minute
02/20/2016 16:15:51 mlat(698): Aircraft: 16 of 28 Mode S, 29 of 63 ADS-B used

mutability, 2016-02-20:
What do you mean by "stalls reading directly from airspy_adsb"?

mutability, 2016-02-20:
Note that the server status: is just a snapshot of the latest status reported; it only tells you something about the approx 30 second period before the report.

touil, 2016-02-20:
Is there any mechanism to compensate for packet losses? This can happen frequently with non real time kernels in tight hardware like the RPI.

jester-xbmc, 2016-02-20:
airspy_adsb -> fa-mlat-client stops "working" after about 5 minutes and server status gives clock unstable, stop working meaning to packets are given back to me (i'm monitoring using -mlatResultsFormat "basestation,listen,31003" to VRS)

with airspy_adsb -> modesmixer -> fa-mlat-client it keeps working, but still gives clock unstable

touil, 2016-02-20:
@mutability Need an airspy? ;-)

mutability, 2016-02-20:
@jester-xbmc: that's probably just luck more than anything. The server tracks how "bad" the instability of a receiver is; over time if a receiver consistently produces unreliable results then it will trust it less and less; past a certain point it stops requesting data for mlat entirely (it keeps taking ADS-B data so it can check if the synchronization fixes itself). There is hysteresis there so once the receiver is inhibited like this, it has to produce good results for quite a long time before it gets turned back on again. Once a receiver is inhibited it's not contributing to mlat so it stops receiving results.

mutability, 2016-02-20:
@touil: The server relies on the timestamp sequence being a continuous sequence, and being approximately the same frequency in the short term. If you drop samples and do not adjust your timestamp to compensate, then that looks to the server like the clock stopped for a while. It will handle this, after enough out of range values are seen it decides they were not just spurious outliers and starts using the new values, but that also flags the clock as unstable. If it happens often enough then the server stops using the receiver's data because it can't be sure the clock is predictable.

dump1090 has a couple of threads: a read thread that's doing the basic USB work via librtlsdr, and the main thread that's running the demodulator etc. Most of the CPU goes in the main thread. They communicate via a fixed-side FIFO of sample buffers. If the read thread needs to hand off a new sample buffer but finds the FIFO is full, then it drops the buffer, but importantly *it still increments the timestamp*. So in the "mild overload" case where the demodulator is not keeping up, but the read thread still gets enough CPU to at least copy the data out from USB, this avoids having the dropped buffer cause a jump in the timestamps. Once things get overloaded to the point where the read thread can't even keep up with USB, that doesn't help though.

See https://github.com/mutability/dump1090/blob/master/dump1090.c#L438 (which notices the FIFO is full and starts dropping data until it's at least half-empty) and https://github.com/mutability/dump1090/blob/master/dump1090.c#L450 ("outbuf->dropped" is the number of samples that were dropped between the last buffer that was processed and the current buffer)

Does airspy_adsb proactively drop samples when it finds it is overloaded? If so, does it adjust the timestamp for the missed samples?

touil, 2016-02-20:
Samples can be dropped in the RTL chip as well. Not whole buffers, but a few samples occasionally when the host computer fails to process the transfer for any given reason. These drops are much harder to track then the ones in the driver.
For airspy, we have counters both in the device side and the host side. It just requires some work.

mutability, 2016-02-20:
Sure, I see exactly that behaviour where small numbers of samples are dropped in bursts (on the order of a few tens or hundreds of samples). There are a few long threads on the flightaware forums about it; it's the #1 cause of clock instability and it is very dependent on the exact setup (the particular dongle, Pi, USB cabling, ambient temperature..)

In practice it is infrequent enough that usually it only makes the receiver drop out for 30 seconds at a time, infrequently. The only case I've seen it be a real problem is with someone with a really long cheap USB cable.

However, this is not the behaviour I see with the airspy data: the airspy receiver is jumping by 100s of _milliseconds_ (see my comment above), that's got to be a whole sample buffer or something similar, not just a failed USB transfer.

touil, 2016-02-20:
That's normal. I didn't implement them yet.

jester-xbmc, 2016-02-20:
Just for reference: its a pi2, with a fresh jessie light OS, running just the airspy, powersupply is official raspberry one, standard 1 meter USB cable that came with the airspy, added one extra clamp ferrite, temp. around 19 C, same setup runs fine with rtl-sdr.

mutability, 2016-02-20:
I took a look at todays logs and it's currently seeing quite a lot of 4.9ms jumps which is different to what it was doing before:

```
20160220 02:09:33.485 master INFO     clocksync            fAT-0024d692b410_000172:fAT-b827eb500a97_000216: fAT-b827eb500a97_000216 clock was 4915.3us slower than predicted, drift error 0.1ppm
20160220 03:02:54.682 master INFO     clocksync            fAT-000c29d5f76e_000071:fAT-b827eb500a97_000216: fAT-b827eb500a97_000216 clock was 4914.9us slower than predicted, drift error 0.2ppm
20160220 03:03:15.250 master INFO     clocktrack           Inhibiting receiver fAT-b827eb500a97_000216 (stage 1: don't use mlat data)
20160220 03:07:45.237 master INFO     clocktrack           Uninhibiting receiver fAT-b827eb500a97_000216
20160220 03:10:20.460 master INFO     clocksync            fAT-000c29d5f76e_000071:fAT-b827eb500a97_000216: fAT-b827eb500a97_000216 clock was 4915.5us slower than predicted, drift error 0.2ppm
20160220 03:10:47.677 master INFO     clocktrack           Inhibiting receiver fAT-b827eb500a97_000216 (stage 1: don't use mlat data)
20160220 03:17:55.623 master INFO     clocktrack           Uninhibiting receiver fAT-b827eb500a97_000216
20160220 03:47:44.515 master INFO     clocksync            fAT-000c29d5f76e_000071:fAT-b827eb500a97_000216: fAT-b827eb500a97_000216 clock was 4915.1us slower than predicted, drift error 0.2ppm
20160220 03:48:00.665 master INFO     clocktrack           Inhibiting receiver fAT-b827eb500a97_000216 (stage 1: don't use mlat data)
20160220 03:50:05.100 master INFO     clocktrack           Uninhibiting receiver fAT-b827eb500a97_000216
20160220 03:50:24.153 master INFO     clocksync            fAT-063e380017dc_000102:fAT-b827eb500a97_000216: fAT-b827eb500a97_000216 clock was 4915.2us slower than predicted, drift error -0.5ppm
20160220 03:50:36.257 master INFO     clocktrack           Inhibiting receiver fAT-b827eb500a97_000216 (stage 1: don't use mlat data)
20160220 03:57:21.593 master INFO     clocktrack           Inhibiting receiver fAT-b827eb500a97_000216 (stage 2: disable mlat data feed)
```

(and it didn't recover until a reconnect)

Anything magic about that 4915us value? (It is a little under 100k samples at 20MHz)

touil, 2016-02-20:
OKI. Let's give it a try.

`static int rx_callback(airspy_transfer_t* transfer)
{
	adsb_parser_advance(transfer->ctx, transfer->dropped_samples);
	adsb_parser_process(transfer->ctx, transfer->samples, transfer->sample_count);
	return do_exit;
}`

touil, 2016-02-20:
4.9ms is what I call an improvement :-) I reworked all the threading yesterday.

jester-xbmc, 2016-02-20:
@mutability new version up and running, shows promise, can you check on server ?

02/20/2016 17:45:11 1882 msgs recv'd from airspy_adsb (1716 in last 5m); 1262 msgs sent to FlightAware
02/20/2016 17:50:11 3767 msgs recv'd from airspy_adsb (1885 in last 5m); 3147 msgs sent to FlightAware
02/20/2016 17:55:11 5765 msgs recv'd from airspy_adsb (1998 in last 5m); 5145 msgs sent to FlightAware
02/20/2016 17:56:33 mlat(758): Receiver status: connected
02/20/2016 17:56:33 mlat(758): Server status:   synchronized with 119 nearby receivers
02/20/2016 17:56:33 mlat(758): Receiver:  936.8 msg/s received     18.5kB/s from receiver
02/20/2016 17:56:33 mlat(758): Server:      0.6 kB/s from server    0.0kB/s TCP to server   2.4kB/s UDP to server
02/20/2016 17:56:33 mlat(758): Results:  292.3 positions/minute
02/20/2016 17:56:33 mlat(758): Aircraft: 19 of 45 Mode S, 40 of 95 ADS-B used

mutability, 2016-02-20:
It is stable enough that it works, but I still see it dropping multiples of 4.9ms from time to time:

```
20160220 18:30:38.973 master INFO     clocksync            fAT-0024d692b410_000172:fAT-b827eb500a97_000327: fAT-b827eb500a97_000327 clock was 39322.2us slower than predicted, drift error -0.4ppm
20160220 18:35:29.919 master INFO     clocksync            fAT-b827eb500a97_000327:fAT-b827ebba5b45_000160: fAT-b827ebba5b45_000160 clock was 9831.4us slower than predicted, drift error -0.1ppm
20160220 19:02:12.211 master INFO     clocksync            fAT-0024d692b410_000172:fAT-b827eb500a97_000327: fAT-b827eb500a97_000327 clock was 4915.3us faster than predicted, drift error -0.1ppm
```

Also it seems to jump in both directions sometimes (in this case, one of the messages used for sync was 4.9ms early, so the server adjusted the offset, but then the next sync point with the same peer receiver using the adjusted offset was 4.9ms late):

```
20160220 18:51:32.109 master INFO     clocksync            fAT-b827eb500a97_000327:fAT-b827ebd75fb6_000002: fAT-b827ebd75fb6_000002 clock was 4915.2us slower than predicted, drift error -0.4ppm
20160220 18:51:32.875 master INFO     clocksync            fAT-b827eb500a97_000327:fAT-b827ebd75fb6_000002: fAT-b827ebd75fb6_000002 clock was 4915.3us faster than predicted, drift error -0.5ppm
```

I wonder if it is a problem with sample blocks being processed out of order?

touil, 2016-02-20:
Let's improve that.

jester-xbmc, 2016-02-20:
@mutability new version up and running, could you entertain us with another peek at the server ?

02/20/2016 21:11:00 3049 msgs recv'd from airspy_adsb (903 in last 5m); 2602 msgs sent to FlightAware
02/20/2016 21:12:33 mlat(709): Receiver status: connected
02/20/2016 21:12:33 mlat(709): Server status:   synchronized with 118 nearby receivers
02/20/2016 21:12:33 mlat(709): Receiver:  508.3 msg/s received     10.0kB/s from receiver
02/20/2016 21:12:33 mlat(709): Server:      0.1 kB/s from server    0.0kB/s TCP to server   1.2kB/s UDP to server
02/20/2016 21:12:33 mlat(709): Results:  55.3 positions/minute
02/20/2016 21:12:33 mlat(709): Aircraft: 6 of 11 Mode S, 30 of 49 ADS-B used

mutability, 2016-02-20:
Mostly OK, a couple of 39.3ms (== 4.9ms*8) glitches:

```
20160220 21:03:44.474 master INFO     clocksync            fAT-063e380017dc_000102:fAT-b827eb500a97_000346: fAT-b827eb500a97_000346 clock was 39321.4us slower than predicted, drift error 0.4ppm 
(... same offset versus lots of other receivers ...)
20160220 21:24:47.143 master INFO     clocksync            fAT-b827eb500a97_000346:fAT-b827eb6d382a_000035: fAT-b827eb6d382a_000035 clock was 39321.6us faster than predicted, drift error -0.1ppm
(... same offset versus lots of other receivers ...)
```

mutability, 2016-02-20:
From a quick look:

You unlock the consumer_mp mutex in consumer_threadproc after decreasing the number of received buffers but before you have copied out the data you just grabbed from the FIFO, around here: https://github.com/airspy/host/blob/master/libairspy/src/airspy.c#L359

But I don't see what prevents the USB callback from reusing that buffer before you've done that copy. If the USB callback does get in first, then you end up processing samples that are FIFO-size samples "in the future" from where you think you are?

Delaying the change to received_buffer_count until after you've done the copy might work (I don't know offhand if it works out cheaper to release and reobtain the lock, or to hold the lock during the copy)

touil, 2016-02-20:
This? https://github.com/airspy/host/blob/master/libairspy/src/airspy.c#L442

mutability, 2016-02-20:
But you release that lock here, before you have done the copy: https://github.com/airspy/host/blob/master/libairspy/src/airspy.c#L361

touil, 2016-02-20:
You mean this copy? https://github.com/airspy/host/blob/master/libairspy/src/airspy.c#L357

[91 more replies]

closed 2016-02-26 by jester-xbmc