flightaware/piaware #32: Local Map fails to Load

jpere039, 2016-07-01:
[image]
[image]

Hi,
Not sure if you are aware but it looks like there is a bug with the local map display. I looked up the error code and it seems like it may be related to google changing their map API?

Thanks,
Joe

mutability, 2016-07-01:
Yeah, it is a change that Google made, they now require an API key which doesn't really work for this case where installs are widely distributed; we're in the process of looking at fixes.

jpere039, 2016-07-01:
cool! If I see anything I'll be sure to let you know.

mutability, 2016-08-19:
This was fixed with the 3.0.3 releases with dump1090-fa using OL3 / OSM.

closed 2016-08-19 by mutability