flightaware/piaware #34: Make run into error

AndiCui, 2016-10-06:
```
make -C package install
make[1]: Entering directory '/home/pi/piaware-master/package'
echo 'pkg_mkIndex .' | 
/bin/sh: 1: Syntax error: end of file unexpected
Makefile:11: recipe for target 'pkgIndex.tcl' failed
make[1]: *** [pkgIndex.tcl] Error 2
make[1]: Leaving directory '/home/pi/piaware-master/package'
Makefile:18: recipe for target 'install' failed
make: *** [install] Error 2
```

mutability, 2016-10-06:
This means you do not have tcl installed (no tclsh in your path); having tcl installed is a build/install prerequisite.

closed 2016-10-06 by mutability

AndiCui, 2016-10-06:
Installed tcl

```
make -C package install
make[1]: Entering directory '/home/pi/piaware/package'
install -d /usr/lib/piaware_packages /usr/lib/piaware_packages/ca
install fa_piaware_config.tcl fa_adept_client.tcl fa_sudo.tcl fa_services.tcl fa_sysinfo.tcl fa_gps.tcl piaware.tcl tryfinallyshim.tcl pkgIndex.tcl /usr/lib/piaware_packages
install ca/*.pem /usr/lib/piaware_packages/ca
c_rehash /usr/lib/piaware_packages/ca
Doing /usr/lib/piaware_packages/ca
flightaware-root.cert.pem => 528167f9.0
flightaware-root.cert.pem => 0de7d325.0
flightaware-ca.cert.pem => f1308d16.0
flightaware-ca.cert.pem => c3719073.0
make -C helpers install
make[2]: Entering directory '/home/pi/piaware/package/helpers'
cc  -Wall -Werror -O  -o droproot droproot.c
install -d /usr/lib/piaware_packages/helpers
install -m 0755 update-piaware-config droproot /usr/lib/piaware_packages/helpers
make[2]: Leaving directory '/home/pi/piaware/package/helpers'
make[1]: Leaving directory '/home/pi/piaware/package'
make -C programs/piaware install
make[1]: Entering directory '/home/pi/piaware/programs/piaware'
install -d /usr/bin /usr/lib/piaware
install -m 0755  /usr/bin/piaware
install: missing destination file operand after ‘/usr/bin/piaware’
Try 'install --help' for more information.
Makefile:14: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/home/pi/piaware/programs/piaware'
Makefile:18: recipe for target 'install' failed
make: *** [install] Error 2
```

mutability, 2016-10-06:
You haven't told it where tcllauncher is.

I suggest you use piaware_builder like the readme suggests.