flightaware/piaware #35: Certificate validation with adept server failed

BugM-eNot, 2017-04-29:
Hello,
I recently fired up my ads-b receiver, in the past I fed my data succesfully to FA, however now I cannot establish a connection with the server, probably because I am getting this message: 

`piaware[2276]: Certificate validation with adept server at piaware.flightaware.com/1200 failed: required field 'subject' is missing
`

Is something wrong with my configuration?

I'm running piaware-git on ArchLinux.

mutability, 2017-04-29:
What's the output from this?

```
openssl s_client -connect piaware.flightaware.com:1200 -showcerts -CAfile /usr/lib/piaware_packages/ca/flightaware-root.cert.pem
```

BugM-eNot, 2017-04-29:
Here is it (see the attachment).
[output.txt](https://github.com/flightaware/piaware/files/966318/output.txt)

mutability, 2017-04-29:
Hm, that seems fine.

Can you uncomment line 236 of /usr/lib/piaware_packages/fa_adept_client.tcl and see what it logs when piaware tries to connect?

```
                # obtain information about the TLS session we negotiated
                set tlsStatus [::tls::status $sock]
                logger "TLS status: $tlsStatus"     ;### uncomment this line
```

BugM-eNot, 2017-04-29:
Here is the new line in the debug log:

`2017-04-29 22:40:36Z Connecting to FlightAware adept server at piaware.flightaware.com/1200
2017-04-29 22:40:36Z Connection with adept server at piaware.flightaware.com/1200 established
2017-04-29 22:40:36Z TLS handshake with adept server at piaware.flightaware.com/1200 completed
2017-04-29 22:40:36Z TLS status: sbits 0
2017-04-29 22:40:36Z Certificate validation with adept server at piaware.flightaware.com/1200 failed: required field 'subject' is missing
2017-04-29 22:40:36Z reconnecting in 51 seconds...
`

mutability, 2017-04-29:
That is pretty weird. It may be a problem with Arch's tcltls package. The TLS handshake says it is succeeding, but then returns no useful info about the negotiated connection.

BugM-eNot, 2017-04-30:
Thanks for your answer @mutability. So, should I contact the maintainer referring to this issue? 
The packages are available here:
- https://www.archlinux.org/packages/?q=tcltls
- https://archlinuxarm.org/packages/?q=tcltls

mutability, 2017-04-30:
I built a standalone testcase, can you try this:

https://gist.github.com/mutability/8cc5e34aa57e262c23bc816a76a068ed

Normal output looks like this:

```
$ ./tls.tcl 
establishing connection
importing socket
handshaking
callback: info sock115f688 handshake start {before/connect initialization}
callback: info sock115f688 connect loop {before/connect initialization}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect exit {unknown state}
handshake: returned 0
handshaking
callback: info sock115f688 connect loop {unknown state}
callback: verify sock115f688 2 {sha1_hash B69ABB0BF41433F4E27434BF6628CE1EA1CAA704 subject {CN=FlightAware Root,OU=Operations,O=FlightAware LLC,L=Houston,ST=TX,C=US} issuer {CN=FlightAware Root,OU=Operations,O=FlightAware LLC,L=Houston,ST=TX,C=US} notBefore {Dec  9 16:50:04 2015 GMT} notAfter {Dec  4 16:50:04 2035 GMT} serial A9FE756D9E6B94B4} 1 {}
callback: verify sock115f688 1 {sha1_hash D191B12012733E764CA27D5DFB88336499C9C7C1 subject {CN=FlightAware Intermediate,OU=Operations,O=FlightAware LLC,ST=TX,C=US} issuer {CN=FlightAware Root,OU=Operations,O=FlightAware LLC,L=Houston,ST=TX,C=US} notBefore {Dec  9 16:59:51 2015 GMT} notAfter {Dec  6 16:59:51 2025 GMT} serial 2048} 1 {}
callback: verify sock115f688 0 {sha1_hash ED69EEEF3FEE10E7DDD1BEC93C10EDCDF88989A1 subject {CN=*.flightaware.com,OU=Operations,O=FlightAware LLC,L=Houston,ST=Texas,C=US} issuer {CN=FlightAware Intermediate,OU=Operations,O=FlightAware LLC,ST=TX,C=US} notBefore {Dec 11 16:33:41 2015 GMT} notAfter {Dec 11 16:33:41 2019 GMT} serial 2049} 1 {}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect exit {unknown state}
handshake: returned 0
handshaking
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 connect loop {unknown state}
callback: info sock115f688 handshake done {SSL negotiation finished successfully}
callback: info sock115f688 connect exit {SSL negotiation finished successfully}
handshake: returned 1
tls status: sha1_hash ED69EEEF3FEE10E7DDD1BEC93C10EDCDF88989A1 subject {CN=*.flightaware.com,OU=Operations,O=FlightAware LLC,L=Houston,ST=Texas,C=US} issuer {CN=FlightAware Intermediate,OU=Operations,O=FlightAware LLC,ST=TX,C=US} notBefore {Dec 11 16:33:41 2015 GMT} notAfter {Dec 11 16:33:41 2019 GMT} serial 2049 sbits 256 cipher AES256-SHA
callback: info sock115f688 alert write {close notify}
```

BugM-eNot, 2017-04-30:
Definitely something different:

```
$ ./tls.tcl
establishing connection
importing socket
handshaking
callback: info sock1cd2b30 handshake start {before SSL initialization}
callback: info sock1cd2b30 connect loop {before SSL initialization}
callback: info sock1cd2b30 connect loop {SSLv3/TLS write client hello}
callback: info sock1cd2b30 connect exit {SSLv3/TLS write client hello}
handshake: returned 1
tls status: sbits 0
```

mutability, 2017-04-30:
OK, definitely a tcltls/openssl problem then. If you pass that testcase on to the tcltls maintainer they may be able to do something with it.

bovine, 2017-04-30:
TclTLS bugs can be reported here: http://core.tcl.tk/tcltls/reportlist

mutability, 2017-04-30:
I can't reproduce the bug with tcltls 1.6 (on a Pi) or 1.6.7 (Ubuntu, x86_64)
I _can_ reproduce the bug on tcltls 1.7.11 (FreeBSD, amd64 - @bovine, that's on graceful)

mutability, 2017-04-30:
Raised a bug report upstream: http://core.tcl.tk/tcltls/tktview/1367823d51d97f90c7d72a3307b5ba7c8757b500

BugM-eNot, 2017-05-01:
It's already fixed in tcltls 1.7.12 [1], I'm waiting for the update in order to test if the piaware issue is gone.
Thanks @mutability and @bovine.

[1]: http://core.tcl.tk/tcltls/artifact/2e0c92707d929982

anish, 2017-05-01:
Let me know when your testing is done, I'll update the requirements for the
PKGBUILD

BugM-eNot, 2017-05-02:
Test done.
Tcltls 1.7.12 just landed in Arch's repo and seems to fix the server connection issue.
@anish you can update the PKGBUILD.

```
TLS handshake with adept server at piaware.flightaware.com/1200 completed
FlightAware server certificate validated
encrypted session established with FlightAware
logged in to FlightAware
```

closed 2017-05-27 by BugM-eNot