flightaware/piaware #36: piaware changed password in config file by itself

anish, 2017-07-17:
My piaware seemed to have changed it's password from the actual password to 'olivertest' by itself at some point last week in piaware.conf. The username remained unchanged. Any idea what could have happened there ?

mutability, 2017-07-17:
No idea - that string doesn't occur anywhere I can find.

You shouldn't set a username/password anyway though.

mutability, 2017-07-17:
I have an email from you dating back to 08/2016 where you arranged remote access for me and used that password, though, so I'm going to guess it's related to that.

anish, 2017-07-17:
That was tls bug I think. It's weird that it would suddenly pick up that config file, I don't have any recollection of puppet username/passwords in piaware.conf either. Is that an override configuration of some sort ?

mutability, 2017-07-17:
Uh, piaware does nothing with puppet, I don't know what you mean?

mutability, 2017-07-17:
To summarize:
 * flightaware-user / flightaware-password in piaware.conf control piaware login to FA only. They do nothing to system login info
 * They are obsolete and shouldn't be set
 * piaware does not ever modify those settings
 * I suspect you set them at some point in the past and don't remember doing it

anish, 2017-07-17:
err, no puppet. I mean piaware. Sorry, wires got crossed at work.

To recap : I got a feeder offline msg from flightaware. When I login to the rpi, I see the service complain about unable to login. I run piaware-config which complains about perms on /etc/piaware.conf. I look at the file, I see olivertest set there. Removing that line makes piaware work fine again.

The olivertest is probably from when you helped debug the issue, I'm not sure why it suddenly started reading that file. I hadn't touched the rpi in about 4 months at this point.

mutability, 2017-07-17:
It's possible that your piaware has just been connected for ages; I don't see any login attempts in the last month or so, so it was connected at least that long. Everything got disconnected yesterday so that may be why it triggered then.

mutability, 2017-07-17:
On a related note can you fix the arch packaging so that it doesn't restart piaware when piaware exits with a fatal error; this is a common problem I see with arch installs.

anish, 2017-07-18:
@mutability will do

mutability, 2017-07-31:
Missed your edit. The right way is how https://github.com/flightaware/piaware/blob/master/scripts/piaware.service does it:

```
Restart=on-failure
RestartPreventExitStatus=4 6
```

closed 2017-07-31 by mutability

anish, 2017-08-13:
@mutability I actually moved to using that service file some time ago. I don't ship a custom one for arch anymore.