flightaware/piaware #37: Exploring the possibility to run PiAware on FreeBSD

yzgyyang, 2017-07-24:
I am interested in trying PiAware on my Raspberry Pi which is running FreeBSD-arm, and exploring the possibility to port PiAware to FreeBSD. I would like to know if there's a more generic build tool, or the requirements of successfully building a standalone PiAware and the required dependencies. Thanks!

jungle-boogie, 2017-07-24:
You might be happy (and surprised) to know Flight Aware uses FreeBSD to run many components of their system! Check out this edition of the FreeBSD journal regarding Flight Aware:
https://www.freebsdfoundation.org/past-issues/globally-accessible-freebsd

Some really good points are made in there about how FreeBSD can be improved.

It would be awesome to have FreeBSD and piaware running together. I don't see much of an issue with it, as this is mostly a tcl app.

yzgyyang, 2017-07-24:
It's good to know that FlightAware loves my favourite system! :smiley: 
From my understanding, [piaware_builder](https://github.com/flightaware/piaware_builder) is just for identifying the platform and setting up dependencies, and is not necessary for this circumstance, is that correct?
I will try building on FreeBSD then, and will post results and questions here.

jungle-boogie, 2017-07-24:
hi @yzgyyang,

Not 100% certain, but by looking at that repo, that's my guess, too. Looks like it makes the actual 'binary' that would be used on the pi running debian.

mutability, 2017-07-25:
Yes, `piaware_builder` handles assembling the various bits needed when building the debian package (the source is spread over a few repositories)

In terms of what's needed there are three main parts:

1. piaware itself. This is the main tcl driver program. It assumes that it's being started from `tcllauncher` (the `piaware` binary is actually tcllauncher) so you'll want to build that too. https://github.com/flightaware/tcllauncher

2. faup1090. This is a C program that receives the raw messages and turns them into summary reports. It is built from `https://github.com/flightaware/dump1090/` ("make faup1090") and piaware expects it to be in `/usr/lib/piaware/helpers/faup1090`

3. fa-mlat-client. This is a python program from https://github.com/mutability/mlat-client. The Debian packaging uses cxfreeze to build a self-contained binary of this to avoid system python version skew problems, but you don't have to do that. piaware expects it to be in `/usr/lib/piaware/helpers/fa-mlat-client`

I guess you will want to build some rc scripts etc for it too.

There are likely to be debian- or linux-specific quirks in the code itself because it's not usually run on other things; let me know what you run into and I can fix those up.

yzgyyang, 2017-07-25:
Hi @mutability , thanks for the lengthy reply!

> the piaware binary is actually tcllauncher
So both [piaware](https://github.com/flightaware/piaware) and [tcllauncher](https://github.com/flightaware/tcllauncher) needs to be built, is that correct? Also I am not sure how to build piaware (`make`?).

mutability, 2017-07-25:
piaware doesn't have a build step as such, since it's all tcl; you should be able to "make install" to put everything in the right place.

yzgyyang, 2017-07-26:
@mutability I managed to build everything successfully. Some modifications need to be made since FreeBSD doesn't have `iptables` equivalent (and similar issues). Will post the progress here soon.

yzgyyang, 2017-07-31:
> FreeBSD doesn't have iptables equivalent (and similar issues)

In `programs/piaware/helpers.tcl`, most of the Linux web commands are not available in FreeBSD, and since I am not really familiar with Linux and TCL, it could take a while until I figure out the equivalents and have a version of this code that could run on FreeBSD.

mutability, 2017-07-31:
What "linux web commands"?

yzgyyang, 2017-07-31:
Sorry, posted the file wrong, it's all the commands in `package/fa_sysinfo.tcl`.

mutability, 2017-07-31:
Ah, yeah, those will all need reimplementing

closed 2026-03-26 by mutability