flightaware/piaware #38: PiAware - no ADS-B data program seen listening, and nothing in /var/log/piaware

charlesreid1, 2017-08-18:
Hi there,

I've installed piaware using the piaware_builder repository for a Debian wheezy system, successfully. This was using the rtl_sdr repository from osmocom and the dump1090 repository from flightaware. The program installs fine, the startup service is ok, and I have successfully configured my username password so that I am connecting to/logging in to the FlightAware servers. I can see this from running the following command:

```
systemctl status piaware
```

(as per the last instruction in the [piaware repo README](https://github.com/flightaware/piaware). However, at the very end of the output from this command, I see the following message:

```
piaware[1172]: no ADS-B data program seen listening on port 30005 for 123 seconds, next check in 60s
```

I'm not sure what that means (what it means by ADS-B program). Perhaps related (but again, not understanding what this means) is the fact that [https://flightaware.com/adsb/stats/user/charlesreid1](https://flightaware.com/adsb/stats/user/charlesreid1) says "No Such ADS-B Site". I'm able to receive signals just fine, because I see signals from planes when I run:

```
dump1090 --interactive
```

which runs smoothly. Further confounding the issue is the fact that there is no `/var/log/piaware` log file generated. When I check the [piaware claim your receiver](https://flightaware.com/adsb/piaware/claim) site, I see the receiver show up in the list. 

Any suggestions as to how to fix the ADS-B problem, or figure out what's going on with the log file, would be awesome & much appreciated.

charlesreid1, 2017-08-18:
output of `$ piaware-config -show`:

```
flightaware-password <hidden>
flightaware-user <hidden>
mlat-results no
wired-network yes
```

I set mlat results to no because I don't know what multilateration is in the context of FlightAware (a short explanation in the README would be helpful) and am just trying to get this working before turning on any bells and whistles.

dmbaker01, 2017-08-18:
dump1090 (generic) or dump1090-fa (FlightAware version) is the program that connects to your RTL dongle receiver and decodes the ADSB messages (ADSB is latest transponder technology). This is the ADSB program that piaware is looking for.

piaware is the program that sends the decoded message to flightaware by listening on the dump1090 output for plane messages

mlat is for decoding and finding positions for older transponders 
https://flightaware.com/adsb/mlat/

You can't claim a site on FlightAware until you send decoded plane positions to them. If you have piaware-config setup you should be able to use "sudo systemctl restart dump1090-fa". 

You can only connect one dump1090 instance to a dongle. So running multiple copies of dump1090 will return a RTL device not found message because the RTL dongle is busy with the other ADSB program.

charlesreid1, 2017-08-18:
Thanks for the quick response! I have dump1090 installed and working standalone. When I start piaware it does not start dump1090. If I try starting dump1090 in another window, the dump1090 program is printing out information on flights, but when I check the status of piaware it still lists the status of the radio as not connected.

The dump1090 program is not on my `$PATH` or installed as a startup service. The binary is just in the directory where I checked it out (from repo `https://github.com/flightaware/dump1090`). I built it using `make BLADERF=no` and am running it like `./dump1090`.

Is dump1090 maybe talking to a different port than 30005, the port that piaware is listening on? Is there a way to specify a port when starting dump1090?

charlesreid1, 2017-08-18:
The problem was with the way I installed dump1090 (specifically, that I had installed dump1090 and not dump1090-fa, and did not have a startup service associated with it). I revisited the [dump1090 installation instructions](https://github.com/flightaware/dump1090/blob/master/README.md) and installed it as a debian package instead of installing it manually. That put a binary called dump1090-fa on my path, and installed a startup service called dump1090-fa. 

Then, I ran `service dump1090-fa start`, then `service piaware start`. At the end of the `systemctl status piaware` I saw the message "piaware received a message from dump1090-fa!" Success!

Documented the whole procedure on the [charlesreid1 wiki](https://charlesreid1.com/wiki/FlightAware). Thanks again for your clarifying comment.

closed 2017-08-18 by charlesreid1