flightaware/piaware #39: How is the json aircraft database generated?

klara31, 2017-11-23:
In the directory dump1090/public_html/db, a json repository contains the relation between the ICAO code, aircraft registration and type. Before this database was generated, using the scripts in the tools/ directory. However, these tools are obsolete as the source is no longer updated.

So how is this database generated these days? Is it possible to update the database myself? I guess a script is used, can you share this script?

mutability, 2017-11-23:
The tools in https://github.com/flightaware/dump1090/tree/master/tools are still the ones used.

klara31, 2017-11-24:
Ok, but which source database do you use? The one at http://www.virtualradarserver.co.uk/Files/BasicAircraftLookup.sqb.gz <http://www.virtualradarserver.co.uk/Files/BasicAircraftLookup.sqb.gz> is not updated anymore.

mutability, 2017-11-24:
The source data is in the tools directory - vrs.csv.gz & flightaware-*.csv.xz

klara31, 2017-11-24:
Then again: how are these two files generated?

mutability, 2017-11-24:
They are _not_ generated, they are the _source data_.

vrs.csv.xz is the last copy of the VRS data before it stopped being updated.
flightaware-*.csv.xz is an extract from an internal FlightAware database of data that we can redistribute; I updated it periodically for major releases.

closed 2018-09-04 by mutability