flightaware/piaware #40: deprecated netstat

a36233, 2018-03-02:
Please change or add support to ss instead of using deprecated tool netstat.
Or at least change netstat arguments to compressed form and then I could symlink ss to netstat.
Old command: COMMAND=/usr/bin/netstat --program --tcp --wide --all --numeric
New proposed  command: COMMAND=/usr/bin/netstat -p -t -w -a -n
Or simply update to the newer version: /usr/bin/ss -p -t -w -a -n

mutability, 2018-03-02:
What distro are you using that's missing netstat?

a36233, 2018-03-02:
ArchLinux

mutability, 2018-03-02:
Given that I don't build/test for Arch, I'd need a patch for this.

a36233, 2018-03-02:
Other distributions will follow to drop net-tools and move on with iproute2 for shure, including debian based distros.
E.g: https://lists.ubuntu.com/archives/ubuntu-devel/2017-January/039651.html
For now as a workaround I just disabled verifications using netstat.
Two solutions to solve this: 
- verify if netstat (net-tools) exist or ss (iproute2) and split output verifications based on both.
- Change detection method to other tool.

mutability, 2018-03-02:
I'm not against moving to ss and friends _if_

1) you (or someone else) can provide a suitable patch; and
2) you can validate that it doesn't break older systems

(note that piaware already uses iproute2 for some things)

The short term fix for Arch would be for the Arch build to depend on net-tools.