flightaware/piaware #41: IndexError: tuple index out of range (Ubuntu 17.10)

jprochazka, 2018-03-22:
Getting a "tuple index out of range" error when building PiAware on Ubuntu 17.10.

```
pyvenv /home/joe/adsb-receiver/build/piaware_builder/package-jessie/debian/venv --without-pip
WARNING: the pyenv script is deprecated in favour of `python3.6 -m venv`
cd cx_Freeze-4.3.4 && /home/joe/adsb-receiver/build/piaware_builder/package-jessie/debian/venv/bin/python ./setup.py install
Traceback (most recent call last):
  File "./setup.py", line 138, in <module>
    finder = cx_Freeze.ModuleFinder(bootstrap = True)
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 160, in __init__
    self._AddBaseModules()
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 187, in _AddBaseModules
    self.IncludeModule("traceback")
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 687, in IncludeModule
    namespace = namespace)
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 355, in _ImportModule
    deferredImports, namespace = namespace)
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 445, in _InternalImportModule
    parentModule, namespace)
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 511, in _LoadModule
    self._ScanCode(module.code, module, deferredImports)
  File "/home/joe/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 597, in _ScanCode
    arguments.append(co.co_consts[opArg])
IndexError: tuple index out of range
debian/rules:55: recipe for target 'build_cxfreeze' failed
make[1]: *** [build_cxfreeze] Error 1
make[1]: Leaving directory '/home/joe/adsb-receiver/build/piaware_builder/package-jessie'
debian/rules:83: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
```
If you need more output from the build let me know.

mutability, 2018-03-22:
You may need to experiment with updating cxfreeze to a newer version (looks like 5.1.1 is current)

jprochazka, 2018-03-24:
First off sorry to post this within this repository. This appears to be a piaware_builder issue.

I have tried to install the newest version of cx_freeze via pip but it seems to be ignored as if piaware_builder continues to refer to the version it downloads.

mutability, 2018-03-26:
You will need to update debian/rules to use your new version of cxfreeze

mutability, 2018-09-04:
3.6.2 is aware of stretch and uses a newer cxfreeze, which fixes this.

closed 2018-09-04 by mutability