flightaware/piaware #42: PiAware update to accommodate PI3 B+

orionjb2001, 2018-03-26:
I believe the new model can not boot using the current PiAware software, users having to update there card in the model B before using that card in a new 3 B+ and there are other issues arising with Pi Aware.  Any idea on an update being released??

mutability, 2018-03-26:
The PiAware software itself shouldn't care, it should work fine on a 3B+.

The PiAware sdcard image first needs to be updated to stretch, then we can add support. It'll happen at some point.

orionjb2001, 2018-03-26:
Thanks Oliver

bhaal2, 2018-04-22:
I'm getting constant clock unstable issues with the Pi 3B+ I've tried all kinds of clock speed settings, some are more stable than others, but either way it's way less stable than other times.  Is this a sensitivity issue which could be adjusted?

mutability, 2018-04-22:
Is the other hardware, including USB cabling, identical?

bhaal2, 2018-04-22:
Yep, nothing changed, all I did was upgrade from Jessie to Stretch..

bhaal2, 2018-04-24:
It seems like it took a while for the clock to settle down, like 12hrs or more, but now it seems to be ok. Seems like the clock has remained stable since this happened..

mutability, 2018-09-04:
The 3.6.2 sdcard images are now Stretch-based.

closed 2018-09-04 by mutability