flightaware/piaware #43: Getting msgs from pfclient, but no msgs sent to FlightAware

Synoptic2, 2018-05-01:
May 1 01:13:20 raspberrypi piaware[1983]: 40 msgs recv’d from pfclient (7 in last 5m); 0 msgs sent to FlightAware”

the count of msgs received goes up, but nothing get sent.

Some other log 👍 
Apr 30 21:43:29 raspberrypi piaware[7370]: piaware version 3.5.3 is running, process ID 7370
Apr 30 21:43:29 raspberrypi piaware[7370]: your system info is: Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri
Apr 30 21:43:31 raspberrypi piaware[7370]: Connection with adept server at piaware.flightaware.com/1200 established
Apr 30 21:43:31 raspberrypi piaware[7370]: TLS handshake with adept server at piaware.flightaware.com/1200 completed
Apr 30 21:43:31 raspberrypi piaware[7370]: FlightAware server certificate validated
Apr 30 21:43:31 raspberrypi piaware[7370]: encrypted session established with FlightAware

mutability, 2018-05-01:
Please describe your configuration on more detail.

Synoptic2, 2018-05-01:
Raspberry Pi 2
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

PF Client version: 3.7.40
HW : Kinect SDR 1090-Puck

PFclient echoing it's data on port 30054

I have Planeplotter, BaseStation and Fr24feeder connected to port 30054 and they all get tracking information correctly.

mutability, 2018-05-01:
Don't think I have ever tried pulling data via pfclient like that. What
format does port 30054 produce? Can you provide a sample?

Synoptic2, 2018-05-01:
Format is binary SBS3. It's hard to give you a sample.

mutability, 2018-05-01:
piaware doesn't support the binary SBS format; it expects Beast.

Synoptic2, 2018-05-01:
I think it is Beast, but I cannot validate. I could forward you the port and you have a look by connecting to it directly ?

mutability, 2018-05-01:
Sure, mail me the details directly if you like

Synoptic2, 2018-05-01:
Hi,
Let me know if you can connect.

leearmstrong, 2018-05-01:
We echo out the input from the receiver.  In this case with a Puck it is not Beast binary but binary Kinetic format.

Synoptic2, 2018-05-01:
Ah ! thank you. That makes sense.

mutability, 2019-08-19:
Currently no plans to support non-beast-format data sources. modes2mixer may be able to translate if you have a less common format and no options for providing beast natively.

closed 2019-08-19 by mutability