flightaware/piaware #44: Flightaware piaware error

orionjb2001, 2018-06-09:
I have a problem installing Flightaware piaware. Running an Odroid XU4, ambrian bionic. Firstly, I am using Joe's ADSB receiver project scripts. First major error, 1090 Flightaware option is not offered as an OS only 1090 mutability. On installing PiAware flightaware, the following error is occuring;

debian/rules build
dh build --with=systemd
dh_update_autotools_config
debian/rules override_dh_auto_configure
make[1]: Entering directory '/home/pi/adsb-receiver/build/piaware_builder/package-jessie'
debian/check_versions
piaware version 3.5.3 matches changelog version 3.5.3, good!
cd tcllauncher && autoconf -f && ./configure --with-tcl=/usr/lib/tcl8.6
checking for correct TEA configuration... ok (TEA 3.9)
configure: configuring Tcllauncher 1.5
checking for Tcl configuration... found /usr/lib/tcl8.6/tclConfig.sh
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for existence of /usr/lib/tcl8.6/tclConfig.sh... loading
checking platform... unix
configure: --prefix defaulting to TCL_PREFIX /usr
configure: --exec-prefix defaulting to TCL_EXEC_PREFIX /usr
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking whether make sets $(MAKE)... yes
checking for ranlib... ranlib
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking if the compiler understands -pipe... yes
checking whether byte ordering is bigendian... no
checking for sin... no
checking for main in -lieee... no
checking for main in -linet... no
checking net/errno.h usability... no
checking net/errno.h presence... no
checking for net/errno.h... no
checking for connect... yes
checking for gethostbyname... yes
checking dirent.h... yes
checking errno.h usability... yes
checking errno.h presence... yes
checking for errno.h... yes
checking float.h usability... yes
checking float.h presence... yes
checking for float.h... yes
checking values.h usability... yes
checking values.h presence... yes
checking for values.h... yes
checking limits.h usability... yes
checking limits.h presence... yes
checking for limits.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
[... 78 lines cut ...]
File "/home/pi/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 511, in _LoadModule
self._ScanCode(module.code, module, deferredImports)
File "/home/pi/adsb-receiver/build/piaware_builder/package-jessie/cx_Freeze-4.3.4/cx_Freeze/finder.py", line 597, in _ScanCode
arguments.append(co.co_consts[opArg])
IndexError: tuple index out of range
debian/rules:55: recipe for target 'build_cxfreeze' failed
make[1]: *** [build_cxfreeze] Error 1
make[1]: Leaving directory '/home/pi/adsb-receiver/build/piaware_builder/package-jessie'
debian/rules:83: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

Installing the FlightAware PiAware client package...

dpkg: error: cannot access archive '/home/pi/adsb-receiver/build/piaware_builder/piaware_*.deb': No such file or directory

Checking that the FlightAware PiAware client package was installed properly...

INSTALLATION HALTED!
UNABLE TO INSTALL A REQUIRED PACKAGE.
SETUP HAS BEEN TERMINATED!

The package "piaware" could not be installed.

FlightAware PiAware client setup halted.

orionjb2001, 2018-06-11:
Update 24 hours later, no change same error occuring. :(

mutability, 2018-06-11:
This is a duplicate of https://github.com/flightaware/piaware_builder/issues/10

closed 2018-06-11 by mutability