flightaware/piaware #49: Rpi 3 a+

r4ptor121, 2019-01-28:
Im having issues installing piaware on a raspberry pi 3 a+

I followed the guide on flight aware and the rpi 3 a+ gets stuck at boot, it only display the rainbow colored screen and nothing happens.
Did the exact same steps using the exact same image on a rpi 3 b+ and worked flawless.

I then tried to install piaware on the a+ using the packages instead. First i installed raspbian and that worked right out of the gate. Installed all the packaged worked aswell but when i try to access the website of the pi, i get "reserverd for lighttpd".

mutability, 2019-01-28:
The current sdcard image predates the release of the 3A+ so it won't work on a 3A+ at the moment. The plan is to build a new image to support this at some point.

A package install should be fine. Note that this is not identical to a sdcard image; in particular the system configuration is not managed via `piaware-config `and the top-level status webpage isn't installed.

r4ptor121, 2019-01-28:
Thats to bad, the A+ is perfect for having outside and powered by solar cells, a single usb for the flight aware stick and wifi for connecting to the network. 
No risk of lightning strikes taking out my PoE switch or other stuff if you having a bad day :)

mutability, 2019-08-19:
The current sdcard image works on a 3A+.
(Doesn't work on a 4B yet, but that's another story..)

closed 2019-08-19 by mutability