flightaware/piaware #51: buffer overflow detected when IPv6 gets disabled

DenyDarko, 2019-02-19:
Hey there,
Yesterday, I've completely disabled ipv6 on my raspberry (ipv6.disable=1).
This was resulted in the following bug:

*** buffer overflow detected ***: /usr/bin/piaware terminated

Thankfully I've found the simple solution which was to remove any IPv6 related entries from /etc/hosts.

After that, piaware launched.

However, you might want to look into it.

Piaware version: 3.6.3

Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/         Linux

mutability, 2019-02-19:
This is, afaik, a bug in the tcl packaged with raspbian rather than a bug
in piaware per se

yippie78, 2020-10-05:
This will help, ipv6 must be enabled for piaware
https://discussions.flightaware.com/t/problem-with-piaware-3-1-0/19222/10
https://discussions.flightaware.com/t/piaware-3-6-2-buffer-overflow-detected/40089/8