flightaware/piaware #55: faup.tcl depends on itcl-3.4 which causes dependency conflicts on Gentoo

daviessm, 2019-05-16:
On current Gentoo, itcl-3.4 depends on tcl<8.6, but the only available version of tclx, 8.4.1, depends on tcl>8.6.
https://github.com/flightaware/piaware/blob/2da8bd97686d3c072ed60ad5643062528f99f92e/programs/piaware/faup.tcl#L3
Can this requirement be increased or unbounded to also accept itcl>4 which depends on tcl>8.6?

mutability, 2019-05-17:
It can be relaxed, but this does point to a packaging problem in Gentoo.
Itcl3 and itcl4 are effectively different implementations, itcl3 can run on
8.6 just fine and can coexist with itcl4 (not in the same interpreter, but
certainly on the same system); Gentoo's packaging should change to allow
this, much as Debian's does

daviessm, 2019-05-17:
Good to know, I'll check that out and report it to them. Thanks.

closed 2019-05-17 by daviessm

SaturnusDJ, 2019-07-07:
Same in Alpine (my experience), and ArchLinux too. https://aur.archlinux.org/packages/piaware-git/#comment-690453. Would be nice if it could be 'worked around' on Piaware side. Unless trouble is possible.

https://core.tcl-lang.org/itcl/dir?ci=tip claims:

> This version [4.1.2] is the next major release to follow
Itcl 3.4.  This version claims to be script level compatible with Itcl 3.4.

mutability, 2019-07-07:
This was already fixed on the dev branch (see the commit above)

SaturnusDJ, 2019-07-07:
Overlooked it. 🤦‍♂ Thanks.

mutability, 2019-07-07:
FWIW, the original reason for requiring itcl 3 was that there's a performance regression going from itcl3 to itcl4. This doesn't really matter for piaware but it's significant on the server side so we avoid using itcl4 in other tcl code.

commit a7c7397b (2019-05-17): Relax itcl package requirement (fixes #55, #53)