flightaware/piaware #57: Failed to config feeder-id

albertwang1988, 2019-09-30:
Hi, there! I've met a probelm trying to config my feeder-id
When I tried:
sudo piaware-config feeder-id XXXXXXXXXXXXXXXXXXXXXXXXX
It shows: could not set option 'feeder-id' to value 'XXXXXXXXXXXXXXXXXXXXXXXXX' as all config files are read-only
It happens when I try to turn 'allow-auto-updates' to yes

I'm running on Debian Buster with dump1090-fa operating perfectly.

I've tried to chmod every file that looked like a config file related to piaware but that didn't make things any better.
 
So help me.

albertwang1988, 2019-09-30:
I may have a solution:
Create /etc/piaware.config manually, and add this to the file
feeder-id XXXXXXXXXXXXXXXXXXXXX
allow-auto-updates yes
allow-manual-updates yes

Then everything work perfectlly

mutability, 2019-09-30:
How did you install piaware?

albertwang1988, 2019-09-30:
> How did you install piaware?
> What architecture / hardware are you running on?

I clone the github repostories: flightaware/piaware and flightaware/dump1090-fa
install all the dependencies and then build deb packages follow instructions contained
then sudo dpkg -i *.deb and all done.

I'm running on VirtualBox, x86 architecture, Debian Buster. Host is MacOS

mutability, 2019-09-30:
Ok, it sounds like `/etc/piaware.conf` went missing for some reason (it is a conffile installed as part of the package; perhaps you removed it?). piaware-config won't try to create missing files.

mutability, 2019-09-30:
I confirmed that the 3.7.2 package we distribute (the armhf one) does include /etc/piaware.conf as a conffile

albertwang1988, 2019-09-30:
I was running this on a x86_64 architecture. I install and uninstall and reinstall many times but the /etc/piaware.conf never created.

mutability, 2019-09-30:
You can check if it's in the package that you built via:

```sh
$ dpkg-deb -I piaware_3.7.2_armhf.deb conffiles | grep piaware.conf
$ dpkg-deb -c piaware_3.7.2_armhf.deb | grep piaware.conf
```

IIRC if you delete a conffile manually, then that change persists (i.e. dpkg won't re-install a conffile that you deleted) across package reinstalls as dpkg assumes that you knew what you were doing when you deleted it and won't override your changes. Removing a package (without purging it) retains config files, so removing/reinstalling doesn't really tell us anything here - if you'd removed the file at any point in the past then that will persist.

There are various ways to force reinstalling the package-provided config files over the top of local changes (here, the local change is the deletion of `/etc/piaware.conf`) , but the simplest way is probably to entirely remove the piaware package _including conffiles_ with `apt-get remove --purge piaware`. This resets everything back to a blank slate including discarding all your local conffile changes. Then reinstall the piaware package.

mutability, 2019-09-30:
For docs on this behavior see e.g. https://www.debian.org/doc/debian-policy/ap-pkg-conffiles.html section 5.1:

> However, note that dpkg will not replace a conffile that was removed by the user (or by a script). This is necessary because with some programs a missing file produces an effect hard or impossible to achieve in another way, so that a missing file needs to be kept that way if the user did it.

albertwang1988, 2019-09-30:
I didn't delete /etc/piaware.conf manually.
Everytime I uninstall the piaware, I use dpkg -P to remove all the conffiles.
I just purge and reinstall piaware again to check the file

`$ dpkg-deb -I piaware_3.7.2_amd64.deb conffiles | grep piaware.conf`
It showed nothing
`$ dpkg-deb -c piaware_3.7.2_amd64.deb | grep piaware.conf`
it showed:

> -rwxr-xr-x root/root     14472 2019-09-11 21:53 ./usr/bin/piaware-config
-rwxr-xr-x root/root       826 2019-09-11 21:53 ./usr/lib/piaware/config.tcl
drwxr-xr-x root/root         0 2019-09-11 21:53 ./usr/lib/piaware-config/
-rwxr-xr-x root/root      4883 2019-09-11 21:53 ./usr/lib/piaware-config/helpers.tcl
-rwxr-xr-x root/root      1531 2019-09-11 21:53 ./usr/lib/piaware-config/main.tcl
-rwxr-xr-x root/root     31540 2019-09-11 21:53 ./usr/lib/piaware_packages/fa_piaware_config.tcl
-rwxr-xr-x root/root      1840 2019-09-11 21:53 ./usr/lib/piaware_packages/helpers/update-piaware-config
-rw-r--r-- root/root      1332 2019-09-11 21:53 ./usr/share/man/man1/piaware-config.1.gz

So maybe there's something wrong with the package? I really didn't deleted the conffiles, I didn't even know /etc/piaware.conf exists before I looked for troubleshooting in the forum of piaware in flightaware.com

albertwang1988, 2019-09-30:
Oh, I'm running on VirtualBox of a x86_64 architecture, I mistakenlly described it as x86, my bad.

mutability, 2019-09-30:
Ok, that's a package build problem then; I don't know why piaware.conf would be missing from the package.

Might be related to building on Buster, Buster is not really tested yet. Can you give me the exact commands you used to clone & build the package?

albertwang1988, 2019-09-30:
Yep, I doubt it's the Debian Buster that to be blamed.
Here's the command I used to clone and build the package:
`$ git clone https://github.com/flightaware/piaware_builder.git`
`$ sudo ./sensible-build.sh`
`$ cd package-stretch`
`$ sudo dpkg-buildpackage -b`
Right before buildpackage I install all the dependecies manually.
Or, should I Issued this on flightaware/piaware_builder?

mutability, 2019-09-30:
`sensible-build.sh` requires at least one argument, can you doublecheck what you ran?

albertwang1988, 2019-09-30:
Sorry, I forgot that it need the version of Debian to run properlly.
I ran
`$ sudo ./sensible-build.sh stretch`

mutability, 2019-09-30:
The `sudo` is unnecessary, but let me try to reproduce it here with what you ran.

albertwang1988, 2019-09-30:
I’m not sure if that sh file needs to be run as root so I add ‘sudo’ in front of it. 
Hope that you can reproduce it.

mutability, 2019-09-30:
I can't reproduce this on a buster armhf install (I don't have an amd64 install around):

```
$ git clone https://github.com/flightaware/piaware_builder.git
$ cd piaware_builder
$ sudo ./sensible-build.sh stretch
$ cd package-stretch
$ sudo apt-get install debhelper tcl8.6-dev autoconf python3-dev python3-venv dh-systemd libboost-system-dev libboost-program-options-dev libboost-regex-dev libboost-filesystem-dev
$ sudo dpkg-buildpackage -b
$ dpkg-deb -I ../piaware_3.7.2_armhf.deb conffiles | grep piaware.conf
/etc/piaware.conf
/etc/rsyslog.d/piaware.conf
```

albertwang1988, 2019-09-30:
So maybe that's the Buster x86_64 problem.
It's easy to solve since you just create /etc/piaware.con manually and everything is ok.
Should we close this issue? Messages here may give enough help to others having the same problem.
Or maybe there's no one trying to run piaware in a laptop like me :)

mutability, 2019-09-30:
Well, I'd like to understand what's going wrong at least.. There's really nothing that is armhf or amd64 specific in the package build so I don't understand why you get a broken package and I don't.

Can you maybe try a clean rebuild? (i.e. wipe out the while `piaware_builder` directory and go through the build process again)

albertwang1988, 2019-09-30:
OK. I'll try it later, maybe tomorrow? I will post the results ASAP.

albertwang1988, 2019-09-30:
I tried to rebuild it and this time the piaware.conf showed!
`$ dpkg-deb -I piaware_3.7.2_amd64.deb conffiles | grep piaware.conf
/etc/piaware.conf
/etc/rsyslog.d/piaware.conf
`
I'm now cofused but anyway problem solved.

mutability, 2019-09-30:
Ok, going to put this down to gremlins then..

closed 2019-09-30 by mutability