flightaware/piaware #58: Receiver type relay never opens connection to host

RawLiquid, 2020-01-12:
Is the relay type broken?
when I set receiver=other, I see 2 connections opened on the host, and data flows directly to FA.
if I then change the receiver type to relay, I never see any connections get opened on the host, and the piaware process keeps restarting itself due to no data.

This is with receiver= other
Jan 12 11:49:22 raspberrypi piaware[1206]: mlat-client(1225): Receiver status: connected
Jan 12 11:49:22 raspberrypi piaware[1206]: mlat-client(1225): Server status:   not synchronized with any nearby receivers
Jan 12 11:49:22 raspberrypi piaware[1206]: mlat-client(1225): Receiver: 1685.6 msg/s received      429.0 msg/s processed (25%)
Jan 12 11:49:22 raspberrypi piaware[1206]: mlat-client(1225): Server:      0.0 kB/s from server    0.1kB/s TCP to server     0.1kB/s UDP to server
Jan 12 11:49:22 raspberrypi piaware[1206]: mlat-client(1225): Aircraft: 62 of 160 Mode S, 101 of 161 ADS-B used
Jan 12 11:49:53 raspberrypi piaware[1206]: 43972 msgs recv'd from the ADS-B data program at 192.168.3.11/30334 (14890 in last 5m); 43972 msgs sent to FlightAware
Jan 12 11:54:53 raspberrypi piaware[1206]: 58142 msgs recv'd from the ADS-B data program at 192.168.3.11/30334 (14170 in last 5m); 58142 msgs sent to FlightAware
Jan 12 11:59:53 raspberrypi piaware[1206]: 71728 msgs recv'd from the ADS-B data program at 192.168.3.11/30334 (13586 in last 5m); 71728 msgs sent to FlightAware

Changing the receiver to relay results in almost no data.
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Listening for Beast-format results connection on port 30105
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Listening for Extended Basestation-format results connection on port 30106
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Route MTU changed to 1500
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Input connected to localhost:30005
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Detected BEAST format input
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Input format changed to BEAST, 12MHz clock
Jan 12 12:04:07 raspberrypi piaware[468]: mlat-client(912): Beast-format results connection with ::1:30104: connection established
Jan 12 12:04:34 raspberrypi piaware[468]: 3 msgs recv'd from dump1090-fa; 2 msgs sent to FlightAware
Jan 12 12:04:55 raspberrypi piaware[468]: piaware has successfully sent several msgs to FlightAware!

and as I said, the host doesn't show an active connection. Is there something else I must do to get the relay working?

STATUS:

 RECEIVER SOFTWARE: ON [stop, display log]
  RAW FEED [30334]: UP, NUM CONNECTED: 0
  BS  FEED [30003]: UP, NUM CONNECTED: 0

mutability, 2020-01-13:
Are you using the piaware sdcard image? Relay mode is only supported on
that image, as it requires configuration of other services as well as
piaware

mutability, 2020-01-13:
Specifically, on a sdcard image in `relay` mode:

* beast-splitter is configured to connect to the remote host for data, listen on port 30005, and establish an outbound data connection to port 30104 (which will be dump1090)
* dump1090 is configured to not listen on port 30005
* piaware connects to beast-splitter on port 30005

If you're not running a sdcard image, then only the piaware part will take effect; you'd need to manually configure the rest if you wanted that sort of setup.

RawLiquid, 2020-01-23:
I had planned to set up a pi with  the latest raspbian and found that certain things aren't quite up to date for it, making compiling difficult.
Is there a link to the image I should start with?  Frankly it would be stellar if it would do both as I am using my second pi for another project.
namely open sdr1 for 1090 traffic, sdr2 for 978 traffic, AND open a tcp connection, sending all of the  data your way. But after toying with it for a while it doesn't seem like that is possible...

mutability, 2020-01-23:
Not sure exactly what you're asking. The 3.8.0 piaware sdcard image (https://flightaware.com/adsb/piaware/build) is up to date with upstream as of the start of the year, and you can apt-get upgrade from there if needed. But if you're going to be putting other stuff on there, I'd suggest starting from a standard Raspbian image and configuring things as you want.

If you just want piaware to connect to an existing source of 1090/978 data that you've set up yourself, then `receiver-type other` and `uat-receiver-type other` are what you're after; they will pull data from whatever is configured by `{uat-,}receiver-{host,port}`. This works with a package install directly. You'll need to separately configure whatever is actually providing that data e.g. dump1090 or whatever.

Note that in this configuration piaware will establish two connections for 1090MHz data (one for the standard feed, one for mlat data); and there is no skyaware integration, so you'll need to get the data to whatever you want to use for visualizing it separately.