flightaware/piaware #59: receiver-port config setting

jonathanfmorgan, 2020-01-28:
Does the receiver-port config setting work? I use a custom port (30054) and I have to go through the source code of each release and replace every instance of 30005 with 30054 to get it to work. Thanks!

mutability, 2020-01-28:
What is your full configuration?

mutability, 2020-01-28:
It works fine for me with:

```
receiver-host                  localhost
receiver-port                  30054
receiver-type                  other
```

```
Jan 28 14:14:26 piaware piaware[8009]: ADS-B data program 'dump1090-fa' is listening on port 30054, so far so good
Jan 28 14:14:26 piaware piaware[8009]: Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30054 --stdout
Jan 28 14:14:26 piaware piaware[8009]: Started faup1090 (pid 8040) to connect to dump1090-fa
```

jonathanfmorgan, 2020-01-28:
Thanks for the reply. I'll re-test and share my config tonight.

jonathanfmorgan, 2020-01-30:
Here's my config:

`
pi@rpi2 ~/piaware_builder/package-jessie $ piaware-config
allow-auto-updates             no                             # value set at /boot/piaware-config.txt:4
allow-manual-updates           yes                            # value set at /boot/piaware-config.txt:7
feeder-id                      xxx # value set at /boot/piaware-config.txt:6
mlat-results-format            beast,listen,30055             # value set at /boot/piaware-config.txt:5
receiver-host                  192.168.101.56                 # value set at /boot/piaware-config.txt:8
receiver-port                  30054                          # value set at /boot/piaware-config.txt:9
receiver-type                  relay                          # value set at /boot/piaware-config.txt:10
`
Here's what I get in the log:

`Jan 29 19:44:48 rpi2 piaware[22431]: piaware version 3.8.0 is running, process ID 22431
Jan 29 19:44:48 rpi2 piaware[22431]: your system info is: Linux rpi2 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux
Jan 29 19:44:49 rpi2 piaware[22431]: Connecting to FlightAware adept server at piaware.flightaware.com/1200
Jan 29 19:44:49 rpi2 piaware[22431]: Connection with adept server at piaware.flightaware.com/1200 established
Jan 29 19:44:50 rpi2 piaware[22431]: TLS handshake with adept server at piaware.flightaware.com/1200 completed
Jan 29 19:44:50 rpi2 piaware[22431]: FlightAware server certificate validated
Jan 29 19:44:50 rpi2 piaware[22431]: encrypted session established with FlightAware
Jan 29 19:44:51 rpi2 piaware[22431]: no ADS-B data program seen listening on port 30005 for 3 seconds, next check in 60s
Jan 29 19:44:51 rpi2 piaware[22431]: UAT support disabled by local configuration setting: uat-receiver-type
Jan 29 19:44:52 rpi2 piaware[22431]: adept reported location: xxx
Jan 29 19:44:52 rpi2 piaware[22431]: logged in to FlightAware as user xxx
Jan 29 19:44:52 rpi2 piaware[22431]: my feeder ID is xxx
Jan 29 19:44:52 rpi2 piaware[22431]: site statistics URL: xxx
Jan 29 19:44:52 rpi2 piaware[22431]: multilateration data requested
Jan 29 19:44:52 rpi2 piaware[22431]: no ADS-B data program is serving on port 30005, not starting multilateration client yet
Jan 29 19:45:22 rpi2 piaware[22431]: 0 msgs recv'd from the ADS-B data program at 192.168.101.56/30054; 0 msgs sent to FlightAware
Jan 29 19:45:52 rpi2 piaware[22431]: no ADS-B data program seen listening on port 30005 for 64 seconds, next check in 60s
Jan 29 19:45:52 rpi2 piaware[22431]: no ADS-B data program is serving on port 30005, not starting multilateration client yet
Jan 29 19:46:52 rpi2 piaware[22431]: no ADS-B data program seen listening on port 30005 for 124 seconds, next check in 60s
Jan 29 19:46:52 rpi2 piaware[22431]: no ADS-B data program is serving on port 30005, not starting multilateration client yet
Jan 29 19:47:52 rpi2 piaware[22431]: no ADS-B data program seen listening on port 30005 for 184 seconds, next check in 60s
Jan 29 19:47:52 rpi2 piaware[22431]: no ADS-B data program is serving on port 30005, not starting multilateration client yet
Jan 29 19:48:52 rpi2 piaware[22431]: no ADS-B data program seen listening on port 30005 for 244 seconds, next check in 60s
Jan 29 19:48:52 rpi2 piaware[22431]: no ADS-B data program is serving on port 30005, not starting multilateration client yet
`
I don't know where the port 30005 is coming from.

My setup is ModeS Beast --USB--> Planefinder.net client --passthrough--> port 30054 to piaware

jonathanfmorgan, 2020-01-30:
When I do the code replace of 30005 for 30054:

Jan 29 20:06:36 rpi2 piaware[26163]: piaware version 3.8.0 is running, process ID 26163
Jan 29 20:06:36 rpi2 piaware[26163]: your system info is: Linux rpi2 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux
Jan 29 20:06:37 rpi2 piaware[26163]: Connecting to FlightAware adept server at piaware.flightaware.com/1200
Jan 29 20:06:38 rpi2 piaware[26163]: Connection with adept server at piaware.flightaware.com/1200 established
Jan 29 20:06:38 rpi2 piaware[26163]: TLS handshake with adept server at piaware.flightaware.com/1200 completed
Jan 29 20:06:38 rpi2 piaware[26163]: FlightAware server certificate validated
Jan 29 20:06:38 rpi2 piaware[26163]: encrypted session established with FlightAware
Jan 29 20:06:40 rpi2 piaware[26163]: ADS-B data program 'pfclient' is listening on port 30054, so far so good
Jan 29 20:06:40 rpi2 piaware[26163]: Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30054 --stdout --lat xxx --lon xxx
Jan 29 20:06:40 rpi2 piaware[26163]: Started faup1090 (pid 26509) to connect to pfclient
Jan 29 20:06:40 rpi2 piaware[26163]: UAT support disabled by local configuration setting: uat-receiver-type
Jan 29 20:06:40 rpi2 piaware[26163]: adept reported location: xxx
Jan 29 20:06:40 rpi2 piaware[26163]: logged in to FlightAware as user xxx
Jan 29 20:06:40 rpi2 piaware[26163]: my feeder ID is xxx
Jan 29 20:06:40 rpi2 piaware[26163]: site statistics URL: xxx
Jan 29 20:06:40 rpi2 piaware[26163]: multilateration data requested
Jan 29 20:06:40 rpi2 piaware[26163]: Starting multilateration client: /usr/lib/piaware/helpers/fa-mlat-client --input-connect localhost:30054 --input-type auto --results beast,listen,30055 --ud$
Jan 29 20:06:40 rpi2 piaware[26163]: piaware received a message from pfclient!
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): fa-mlat-client 0.2.11 starting up
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): Using UDP transport to 70.42.6.224 port 6570
Jan 29 20:06:41 rpi2 piaware[26163]: piaware has successfully sent several msgs to FlightAware!
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): Listening for Beast-format results connection on port 30055
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): Route MTU changed to 1500
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): Connection to localhost:30054 lost: [Errno 111] Connection refused
Jan 29 20:06:41 rpi2 piaware[26163]: mlat-client(26520): Reconnecting in 0.5 seconds
Jan 29 20:06:42 rpi2 piaware[26163]: mlat-client(26520): Input connected to localhost:30054
Jan 29 20:06:42 rpi2 piaware[26163]: mlat-client(26520): Detected BEAST format input
Jan 29 20:06:42 rpi2 piaware[26163]: mlat-client(26520): Input format changed to BEAST, 12MHz clock
Jan 29 20:06:46 rpi2 piaware[26163]: mlat-client(26520): Accepted Beast-format results connection from ::ffff:192.168.101.61:49649
Jan 29 20:07:10 rpi2 piaware[26163]: 79 msgs recv'd from pfclient; 79 msgs sent to FlightAware

mutability, 2020-02-02:
Is this a piaware sdcard image? `receiver-type relay` requires configuration of other software that is only done on a sdcard image, not a package install. If you are not using the sdcard image, use `receiver-type other` and arrange your own configuration.

jonathanfmorgan, 2020-02-03:
Yes it's a package install and that did the trick.  Thanks!

closed 2020-02-03 by jonathanfmorgan