flightaware/piaware #61: dump1090 build optimisations for debian

nimbius, 2020-02-21:
during dump1090 builds, optimization warnings are thrown.  can these be implemented?

dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/dump1090-fa/usr/bin/dump1090-fa debian/dump1090-fa/usr/bin/view1090-fa were not linked against libtinfo.so.5 (they use none of the library's symbols)
dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/dump1090-fa/usr/bin/dump1090-fa debian/dump1090-fa/usr/bin/view1090-fa were not linked against ld-linux-armhf.so.3 (they use none of the library's symbols)

mutability, 2020-02-21:
These are false positives.

mutability, 2020-02-21:
specifically:

 * ld-linux is the dynamic linker, you can't remove that
 * libtinfo is a transitive dependency of libncurses

closed 2020-02-21 by mutability