flightaware/piaware #67: About feed with pi3 a +

dsukjp767, 2020-10-09:
Hello
I'm using translation because I'm not good at the English.  sorry.
I'm trying to feed using the recently released pi3 a +.  It seems that the cpu performance is better than pi2 and pi3, so basically I'm not dissatisfied, but for some reason I can't feed to flightaware.  I have installed the package and dump1090-fa is working properly and is able to feed other services.  However, even if I try using the top command, piaware is not working and it is not displayed on the flightaware management screen.
Is there any workaround for this?
I'm sorry for the rudimentary question.  Thank you for reading the long text.

eric1tran, 2020-10-09:
Can you enter `piaware-status` and send me the output? 

Also what do your piaware logs say? `tail -f /var/log/piaware.log`

dsukjp767, 2020-10-10:
Thank you for your prompt response.
The status is output as command not found.
The log looks like this:
tail: cannot open '/var/log/piaware.log' for reading: No such file or directory
tail: no files remaining

I have repeated the official procedure from the beginning of rewriting the raspi-os about twice, but this is the situation every time.

eric1tran, 2020-10-10:
I don't think you have piaware installed. 

The simplest solution would be to use the SD card image: http://piaware.flightcdn.com/piaware-sd-card-4.0.img.zip

If you want to go the package install route, following the steps at number 2 here and let me know the output: https://flightaware.com/adsb/piaware/install

dsukjp767, 2020-10-10:
When I tried -sudo apt-get install piaware- again according to the official procedure introduced, it was installed normally and it was also displayed on the management screen.
Thanks to the easy-to-understand advice, I was able to solve it. Thank you very much for your busy schedule.

eric1tran, 2020-10-10:
You're welcome.

closed 2020-10-10 by eric1tran