flightaware/piaware #70: reported internal ip address / docker

mo8Zomo0, 2021-03-15:
Hi!

Piaware reports back the internal IP address and exposes that as a link on the users flightaware page.

https://flightaware.com/adsb/stats/user/ shows the internal IP and port of the local webinterface
under "site information": "Web Interface: View live data (requires local network connection)"
which show something like: http://172.17.0.6:8080/

Especially with using docker this internal address might not even be accessible.

It would be nice if the behaviour could be set / overwritten.

Like behaviour: 
default export any internal address)
manual (configure protocol (http/https), host/ip, port as whatever you like)
auto-docker (get the hosts ip/port if this is even possible

mutability, 2021-03-15:
Here's the existing logic: https://github.com/flightaware/piaware/blob/master/package/fa_sysinfo.tcl#L292
What's the equivalent for detecting a docker environment and extracting the right address?

resuna, 2021-03-15:
I just went over this, and there's three things you can do, all of them "not recommended" or "not reliable".

1. Set the docker hostname to the real hostname instead of a container ID, and do a DNS lookup on that. We do this. Not reliable because it assumes you control the docker.
2. Tunnel it into docker using an environment variable. Not reliable for the same reason.
3. Exposing the docker management API inside the container, which basically gives you root on the host. Not recommended.

mo8Zomo0, 2021-03-16:
As said, an option to define the behaviour and one to define a manual value for the URI could be a fix.

I agree IPv6 would  most likely be the most handy thing, though a question would be even if you 
use v6 in containers if you would expose it there or concentrated at an other point.

resuna, 2021-03-16:
Not doing address translation is kind of a religion for v6 promoters.

Anyway, yeh, just making it some parameter or setting you pass to the program to override the ip address is the way to go. That way it doesn't matter if it's docker or some other NAT/PAT system that's hiding the address from you.

davmonk, 2026-11-16:
I looked into a PR for this, but the right fix depends on the server side, so I'd like to ask which approach you'd accept first.

As far as I can tell, piaware only sends `local_ip` and `local_iface` (in the health message, `programs/piaware/health.tcl:28-29`, and at login, `programs/piaware/login.tcl:65-66`), taken from `ip route get` towards FlightAware. The "Web Interface" link (`http://<ip>:8080/`) is built server-side. So the part that breaks for Docker, a host port that differs from 8080, isn't something piaware can control today.

Two options I can see:

1. **A new override setting sent as its own field,** e.g. `local-web-url`, reported as `local_web_url` and left for the server to use when present. This follows how `force-macaddress` is sent as `forced_mac` (`login.tcl:89`) rather than replacing the detected value. It would handle host, port and scheme, but does nothing until the site uses the field.
2. **A setting that replaces the reported `local_ip`.** This needs no server change, but only fixes the host part, and it changes a value the server may use for other things I can't see.

Would you take (1)? If so, what field name or format should it use? I'm happy to write it, with tests, once I know which side you want it on.