flightaware/piaware #72: Route information, etc.

developerfromjokela, 2021-06-15:
Hello!
This guy has images and route information shown directly on his piaware:
https://flight.f4ipo.fr/

How could I do something like that?

Brummby, 2021-06-15:
The comments in his code are rather amusing... ``

developerfromjokela, 2021-06-16:
That code is in every flight aware instance, not just his one 😀

serge-golovanow, 2021-07-30:
Hello ! I'm "this guy" 😉 I made some custom code in order to display routes and images : 
- https://flight.f4ipo.fr is hosted on a dedicated server, using ApacheHTTPd ; all static content is here, only json files are taken from a piaware instance hosted at home, increasing performance and hidding my IP, https and http/2 are some bonus. Then a reverse_proxy could be enough to get dynamic data, but aircraft.json is in fact a custom PHP page (hidden with url rewriting) : the real aircraft.js is parsed, then some data is added :
- routes and flight operator are taken in StandingData.sqb, from [VirtualRadarServer](http://www.virtualradarserver.co.uk/flightroutes.aspx) : I update it daily (and of course I send updates to [StandingDataMaintenance](https://sdm.virtualradarserver.co.uk/))
- additional data about plane (type, owner, ...) is from BaseStation.sqb, from a VRS instance running at home, with [DataBaseWrite](http://www.virtualradarserver.co.uk/Download.aspx#panel-database-writer) plugin to keep all the data, and a [ADSBHub](https://www.adsbhub.org/howtogetdata.php) receiver : I see as much planes as possible in order to populate the database (presently 518k records !). I just keep VRS running sometimes, then push the database (I also compute it to [mark military planes](http://www.virtualradarserver.co.uk/FlightRoutes.aspx#codeblock-lookup))
- I search for images in Flickr with its API, storing results in a third sqlite db
- planeObject.js and script.js are adapted to parse and display this additional content.

sharkbit, 2022-05-19:
@serge-golovanow Any chance on pushing the changes in your branch up to GIT.   Pretty cool modification!

MaxWinterstein, 2024-05-27:
@serge-golovanow I know we all hate bumps, but - are you still on this? :)

HNJAMeindersma, 2024-07-12:
Not sure if this should be added to instances globally, since it would add an extra dependency and increase traffic to the provider(s) of route information. Making it an addon or additional package would be more appropriate in my opinion.