flightaware/piaware #73: Character Handling: PiAware Config

pgrady, 2021-06-20:
Wifi network contains a value utf-8 character (e-grave which prints as "è"). With this character entered in /boot/piaware-config.txt, resulting wpa-supplicant.conf is "??" instead of the e-grave character. None of the obvious quote patterns (quoting the string, single backslash, double backslash, unicode (u\)) work. Looks like the character handling doesn't support utf-8 (and probably utf-16). Converting the SSID to hex in piaware-config.txt also doesn't work.

mutability, 2021-06-20:
SSID character set encoding is, at best, poorly defined. What encoding does wpa-supplicant expect in the config file?

mutability, 2021-06-20:
I guess we should detect the "looks like a hex string" case and not quote it, though I am wary of breaking things that look like a hex string but actually are ASCII (C0FFEE..)

davmonk, 2026-11-16:
I think the original problem here is fixed on current images. The network config is no longer written as `wpa_supplicant.conf`. piaware-support's `generate_network_config.py` writes a NetworkManager connection file, and when I ran it with `wireless-ssid "Cafè réseau"` in a UTF-8 `piaware-config.txt`, it wrote `ssid=Cafè réseau` correctly (Debian 13, Python 3.13).

It does break if the file isn't saved as UTF-8. Older Windows Notepad saves as Windows-1252 by default, and a single `è` saved that way, even in a comment, stops the config from loading and prevents any network config from being generated. I've filed that as flightaware/piaware-support#26, since the code lives there.

Until that's fixed: save `piaware-config.txt` as UTF-8, which is the default in current Notepad. (Unverified on a real sdcard image.)

davmonk, 2026-11-16:
Correction to my earlier comment: the original problem here is **not** fixed.

piaware's own config code (`package/fa_piaware_config.tcl`, used by `piaware-config`) reads and writes these files with `-encoding ascii`. Running the real read/write path:

- `piaware-config wireless-ssid "Café"` writes `Caf?`.
- With a correctly hand-entered UTF-8 SSID, changing **any other** option with `piaware-config` rewrites the whole file, and the SSID becomes `Caf??`. Each `é` is two bytes, and each becomes a `?`. That's the `??` reported here.

What I tested before was only the network config generator reading a hand-edited file, which handles UTF-8 correctly. #149 switches piaware's config code to UTF-8 and includes tests.