flightaware/piaware #74: Airbus A220 aircraft do not show aircraft type

cemeyer2, 2021-07-08:
On the page that shows the table of aircraft with details like altitude, etc Airbus A220 aircraft show up with no value in the "Aircraft Type" column. 

Clicking on the flight to go to flightaware.com does show that it is an A220, but it does not show up on the piaware UI

cemeyer2, 2021-07-08:
for example, the `DAL` flights here are both A220 aircraft but show no value

[image]

mutability, 2021-07-09:
This is likely to be because the data extraction process used to generate the registration database for SkyAware from FAA data doesn't know about the A220 model numbers (they are listed as, for example, BD-500-1A11 in the FAA registry)

mutability, 2021-07-14:
This is fixed upstream and will filter through when the skyaware static database is updated for the next release

cemeyer2, 2021-10-01:
it looks like they now show up as `BCS1` whereas other airbus aircraft are `A330` or similar. It seems that given this is upstream, this is not something piaware can directly fix, but wanted to get context on the difference

mutability, 2021-10-01:
BCS1 is the correct ICAO type designator for an Airbus A220:

[image]

(it was originally the Bombardier C-Series and was rebranded to Airbus later)

krocheck, 2021-12-08:
You'll also see `BCS3` for the `A220-300`.  This show correctly for me currently, as I saw a few of both types going in and out of DTW the other day.  This can probably be closed.