flightaware/piaware #76: Enhancement request: direct entry of latitude and longitude values when setting location

jaymot123, 2021-09-11:
Those who already know their exact GPS coordinates are still forced to use the map in the settings window to set their location, which may not be as precise as entering known-accurate latitude and longitude values. Please add the ability to directly enter these values if they're known.

eric1tran, 2021-09-11:
Are you talking about your My ADS-B stats page on www.flightaware.com? If so, both of your requests are already possible.

jaymot123, 2021-09-11:
@eric1tran yes. The only selection I see when I click on the gear icon then click manually enter location is to use the map. I find nothing that lets me manually enter my latitude and longitude.

mutability, 2021-09-13:
You can already do this:

[image]

[image]

closed 2021-09-13 by mutability

jaymot123, 2021-09-16:
Aha! I guess I didn't scroll far enough down in the window to see that at the bottom.  (Sorry.)