flightaware/piaware #77: MLAT error

pbhound, 2021-09-29:
I have upgraded my PiAware to 6.0 (new SD card image due to failure on old SD card) the website still shows my station as having PiAware 5.0 vise the PiAware actually stating "welcome to PiAware 6.0!"; but that's another issue.

the real issue I am having is that the MLAT is stating "error" when I have it enabled in the software. is this a known issue or is there something I need to do?

mutability, 2021-09-29:
Have you configured the new sdcard with your old feeder-id, or claimed the new install as a new site and set its location?

pbhound, 2021-09-29:
> 
> 
> Have you configured the new sdcard with your old feeder-id, or claimed the new install as a new site and set its location?

yes, I have. how do I get rid of the previous installation that is still listed on the website?

closed 2021-09-30 by pbhound

mutability, 2021-10-01:
Old sites go away on their own, or I believe there's an option on the stats interface to remove them manually after they've been silent for a while.

Did you resolve the original problem?

pbhound, 2021-10-01:
> 
> 
> Old sites go away on their own, or I believe there's an option on the stats interface to remove them manually after they've been silent for a while.
> 
> Did you resolve the original problem?

I will have to look for the setting. 

as for the MLAT, no its still showing as disabled on the Pi but on the website its saying enabled.