flightaware/piaware #80: Prevent automatic rebooting when no A/Cs are detected or increase check interval

guidocioni, 2021-12-16:
On my Raspberry Pi 4 running Ubuntu I'm having trouble with my RTL-SDR driver: every time the device is opened it cannot be closed otherwise at the next attempt of opening the connection will report a -7 error and the only way to fix this is to unplug and plug the device again.

I know this is an issue with this system as I tried on another Raspberry running Debian and had no issues. Also the same device with an older version of Ubuntu was running fine. I tried everything, don't know how to fix it. I'm getting over it :) 

Now to the piaware issue. 

In case no A/Cs are detected in about an hour the system tries to restart everything 

```
Dec 16 01:15:11 ubuntu piaware[696232]: no new messages received in 3712 seconds, it might just be that there haven't been any aircraft nearby but I'm going to try to restart everything, just in case...
```

This usually is not an issue on a "normal" setup but in my case, as the device is closed an re-opened, causes a failure which I can only fix by physically plugging and unplugging the dongle again...which is a pain in the ____ if I'm not on site.

I read the documentation but I haven't found any way to disable this automatic restart or at least to increase the interval to few hours instead than just one. 
Is this something that can be changed in the configuration?

guidocioni, 2021-12-16:
> Not currently configurable, but I'd take a PR to add an option for this

Can you point me to the relevant part in the code where this interval is set? I'm not familiar with the repo. 

I think the easiest way to fix this would be to put this interval in the configuration instead than hard-coding it.
Disabling the automatic restart is maybe not needed as you can always specify an interval large enough.

mutability, 2021-12-16:
You'd need to add a config option in [fa_piaware_config.tcl](https://github.com/flightaware/piaware/blob/e0f8c3a776741439b9be32376654943707544a63/package/fa_piaware_config.tcl#L1002)

and change the restart logic in [faup.tcl](https://github.com/flightaware/piaware/blob/e0f8c3a776741439b9be32376654943707544a63/programs/piaware/faup.tcl#L239)