flightaware/piaware #81: Enable text compression on lighthttpd

cqueern, 2022-01-13:
Hello FlightAware team, 

Text-based resources should be served with compression (gzip, deflate or brotli) to minimize total network bytes. Would it be possible to enable compression of text resources sent by lighthttpd in an upcoming release? 

[Mod_deflate](https://redmine.lighttpd.net/projects/1/wiki/Docs_ModDeflate) may be the right way to get that done.

This [resource](https://web.dev/uses-text-compression/#how-to-enable-text-compression-on-your-server) suggests brotli should be first choice over gzip. Will leave it to you to decide whether brotli or gzip is more appropriate. 

Thanks for all your hard work.

mutability, 2022-01-13:
Do you have proposed config changes for this?

cqueern, 2022-01-13:
> Do you have proposed config changes for this?

Based on what I see here:

https://redmine.lighttpd.net/projects/1/wiki/Docs_ModDeflate

it looks like a safe place to test mod_deflate might be with the out of the box recommendations:

`deflate.mimetypes = ("text/html", "text/plain", "text/css", "text/javascript", "text/xml")`
`deflate.allowed-encodings = ( "br", "gzip", "deflate" ) # "bzip2" and "zstd" also supported`

mutability, 2022-01-14:
Well, more specifically, I'm interested in things like:

 * are you proposing to compress the dynamically generated json on the fly? How expensive is this?
 * for static content, pre-compressing the data would seem to be a better approach than mod_deflate
 * can you scope this so that it is not global?

The devil is in the details and I'd really need to see a concrete proposal here.

I'd point out that _usually_ skyaware is feeding the local network only, and so bandwidth isn't really an issue.

cqueern, 2022-01-14:
Thanks for the specific questions @mutability.

> are you proposing to compress the dynamically generated json on the fly? How expensive is this?

Good clarification. No, I don't propose json should be compressed. I don't think the proposed configs shared above would affect json.

> for static content, pre-compressing the data would seem to be a better approach than mod_deflate

Not sure I'm familiar with pre-compressing. Are you talking about [minification](https://developers.google.com/web/fundamentals/performance/optimizing-content-efficiency/optimize-encoding-and-transfer#minification_preprocessing_context-specific_optimizations)? If so, I think that's already in place in piaware's heavier js and css and not what I hoped to request in this Issue. 

> can you scope this so that it is not global?

I am not aware of a way to do so (but that doesn't mean there is not one). Is there a specific part of the idea that makes you wonder about that?

> I'd point out that usually skyaware is feeding the local network only, and so bandwidth isn't really an issue.

Understood. I believe it's likely a safe, low level-of-effort request that should make a noticeable impact so thought it was worth exploring.

gstrauss, 2022-01-22:
@mutability wrote:
>> I'd point out that usually skyaware is feeding the local network only, and so bandwidth isn't really an issue.

That point is key.  @cqueern please identity the real/actual (not theoretical) issue you are trying to address.  If client access is on a local high-bandwidth and low-latency network, and if the server device is not network I/O bound, then the benefit to using lighttpd mod_deflate is likely to be marginal, at best.  The content also needs to be compressible and cacheable, and even then will not be re-accessed often by the same client if proper caching headers have been supplied with lighttpd mod_expire.

@mutability wrote:
> are you proposing to compress the dynamically generated json on the fly? How expensive is this?

Not very expensive unless server device is underpowered or overloaded.

> for static content, pre-compressing the data would seem to be a better approach than mod_deflate

`deflate.cache-dir` can be used to configure a location for lighttpd to cache compressed responses.  lighttpd will reuse the already-compressed response as long as the original file has not changed (checked with ETag comparison).  This means that for static resources which change infrequently, the response will already be compressed, and the CPU usage will be minimal.

> can you scope this so that it is not global?

Yes, lighttpd can be configured to scope mod_deflate to parts of the site.  Such configurations are specific to each application.

@cqueern wrote:
> Understood. I believe it's likely a safe, low level-of-effort request that should make a noticeable impact so thought it was worth exploring.

Sounds like you, @cqueern, should do the exploring rather than suggesting someone else to do that work to measure and gauge the benefits.  What evidence can you share to buttress your statement "I believe it's likely a safe, low level-of-effort request that should make a noticeable impact"?  (Focus on data supporting your conclusion "should make a noticeable impact")  Do you have a waterfall showing that the download of static assets are the bottleneck and are slowing page loads, and that enable lighttpd mod_deflate will reduce this bottleneck?

cqueern, 2022-01-31:
@gstrauss appreciate all the feedback. I definitely don't have much lighttpd experience and it looks like you're familiar with the nuances. 

> What evidence can you share to buttress your statement "I believe it's likely a safe, low level-of-effort request that should make a noticeable impact"? (Focus on data supporting your conclusion [...]

I don't have a waterfall, sorry. My experience is based on implementing text compression on apache webserver a few times and nginx once or twice. It was straightforward, low level of effort, and impactful which lead to my hope that it might be the same here too. 

Defer to you and @mutability whether it makes sense here.

gstrauss, 2022-01-31:
@cqueern It sounds like you have not made any measurements and that you do not have any piaware performance data (running behind any of Apache or nginx or lighttpd).

Yes, as I already said above, lighttpd.conf can be configured to enable mod_deflate on certain mimetypes for certain parts of a site (application-specific) in lighttpd.conf.  Please refer to the [lighttpd documentation](https://wiki.lighttpd.net/) for how to do so.

Whether or not it is worthwhile to do so for piaware is unknown (at least to me) because no data has been presented for review.

@cqueern wrote:
> Defer to you and @mutability whether it makes sense here.

Repeating what I said above:
> Sounds like you, @cqueern, should do the exploring rather than suggesting someone else to do that work to measure and gauge the benefits.

cqueern, 2022-02-08:
>Sounds like you, @cqueern, should do the exploring rather than suggesting someone else to do that work to measure and gauge the benefits.

I'm still working on the analysis and would ask that this Issue be kept open in the meantime. Thank you.

cqueern, 2022-02-19:
Hello Team, I've measured performance data that gauges the benefits of enabling text compression on lighttpd. 

### Enabling text compression step-by-step ###

1. Based on advice [here](https://discourse.pi-hole.net/t/mod-deflate-so-module-is-missing-from-lighttpd/42781/2) I ran the following:

`sudo apt-get install lighttpd-mod-deflate`

2. I edited the `lighttpd.conf` file found in the `/etc/lighttpd` folder by adding `"mod_deflate",` in the `server.modules` section at the very top.

3. I edited the `lighttpd.conf` file found in the `/etc/lighttpd` folder by appending this to the very bottom of the file
```
#deflate.mimetypes = ("text/")       # prefix matches all text/* Content-Type responses
deflate.mimetypes = ("text/html", "text/plain", "text/css", "text/javascript", "text/xml")
deflate.allowed-encodings = ( "br", "gzip", "deflate" ) # "bzip2" and "zstd" also supported
```

4. I rebooted the raspberry pi.

#### Example HTTP Response Headers with and without compression ####
Without compression enabled  | With compression enabled
------------- | -------------
[image]  | [image]

### Performance Testing ###

I used [Lighthouse 9.1.0](https://github.com/GoogleChrome/lighthouse/releases/tag/v9.1.0) in Chrome 98.0.4758.102 and captured the overall [Performance](https://web.dev/performance-scoring/) score of the [SkyAware 7.1](https://github.com/flightaware/dump1090) map found at http://_local ip address_/skyaware/  twenty times with and without compression enabled in lighttpd. There was no ADS-B antenna connected to the raspberry pis used for testing as I only have one and that's [in use](https://flightaware.com/adsb/stats/user/cqu) right now. 

For reference, Lighthouse Performance scores range from 0-100 and [can be interpreted](https://web.dev/performance-scoring/#color-coding) as such:

- 0 to 49 (red): Poor
- 50 to 89 (orange): Needs Improvement
- 90 to 100 (green): Good

Lighthouse Performance Score Metric  | Without compression enabled (N=20) | With compression enabled (N=20)
------------- | ------------- | -------------
Mean | 67.6  | 77.0
Standard Deviation  | 7.34  | 8.58

I used [this site](https://www.socscistatistics.com/tests/studentttest/default2.aspx) to test for statistical significance. In a two-tailed T-test, the t-value is 3.72119. The p-value is .00064. The result is significant at p < .01.

### Conclusion  ###

There appears to be worthwhile performance improvement on FlightAware when enabling compression on lighttpd. Recommend that the project maintainers consider enabling compression on lighttpd.

gstrauss, 2022-02-19:
@cqueern thank you for sharing data to support this.

On 13 Jan, @mutability posted some questions, and I posted some responses to both of you.  I would suggest that you try scoping the config, as requested by @mutability.  I am not familiar with the piaware file layout, but I will provide this example: if static resources are located under `/static/`, then you might use something like the following:
`$HTTP["url"] =~ "^/static/" { deflate.cache-dir = "/dev/shm/lighttpd-cache" }`  With this config -- and in addition to your proposed mod_deflate config -- then as static resources are requested, the compressed response will be stored in the cache and then reused for future requests.  /dev/shm/ is an in-memory filesystem on Linux and is limited in size.  Is this appropriate for piaware?  An alternative location should be used (such as some place under `/var/cache`) if the compressed resources would consume too much space (and memory).  If created under `/var/cache`, then some root process needs to create the directory and assign permissions to the user under which the web server runs, and a periodic job should clean up files.

cqueern, 2022-02-27:
Hello Team. 

I think one concern is that by not caching the compressed versions of the text resources on the web server, we'll be making lighttpd work harder than is necessary because it will have to compress each text resource each time before sending it to the browser.

Beyond the original request to enable text compression on lighttpd, I considered **enabling lighttpd to tell the browser to cache static resources** (compressed text resources and others like images) so that lighttpd doesn't have to compress each text resource each time before sending it to the browser. Compression of text resources and [caching of static resources](https://web.dev/uses-long-cache-ttl/#how-to-cache-static-resources-using-http-caching) are fundamental techniques for front-end browser performance so I investigated a bit. Here's what I did to try this out...

## Enabling caching of static resources step-by-step

Note: the changes described below are in addition to the changes [described above](https://github.com/flightaware/piaware/issues/81#issuecomment-1046105840) to enable text compression. 

1. Based on advice [here](https://www.cyberciti.biz/faq/lighttpd-send-cache-control-maxage-headers-of-staticfiles/) and [here](https://viewsby.wordpress.com/2012/10/11/lighttpd-add-expire-and-cache-control-headers/) I edited the `lighttpd.conf` file found in the `/etc/lighttpd` folder by adding `"mod_setenv",` in the `server.modules` section at the very top.
2. I edited the `lighttpd.conf` file found in the `/etc/lighttpd` folder by appending this to the very bottom of the file
```
$HTTP["url"] =~ "\.(jpg|gif|png|css|js|txt|ico)$" {
    setenv.add-response-header = ( "Cache-Control" => "max-age=31536000" )
}
```
3. I rebooted the raspberry pi.

#### Example HTTP Response Headers with and without caching of static resources ####
Without caching of static resources enabled  | with caching of static resources enabled
------------- | -------------
[image] | [image] 

### Performance Testing ###

I used [Lighthouse 9.1.0](https://github.com/GoogleChrome/lighthouse/releases/tag/v9.1.0) in Chrome 98.0.4758.109 and captured the overall [Performance](https://web.dev/performance-scoring/) score of the [SkyAware 7.1](https://github.com/flightaware/dump1090) map found at http://_local ip address_/skyaware/ , twenty times comparing two different lighttpd configurations:
1. without compression and without caching of static resources
2. with compression and with caching of static resources

There was no ADS-B antenna connected to the raspberry pis used for testing as I only have one and that's [in use](https://flightaware.com/adsb/stats/user/cqu) right now. 

For reference, Lighthouse Performance scores range from 0-100 and [can be interpreted](https://web.dev/performance-scoring/#color-coding) as such:

- 0 to 49 (red): Poor
- 50 to 89 (orange): Needs Improvement
- 90 to 100 (green): Good

Lighthouse Performance Score Metric  | without compression and without caching of static resources enabled (N=20) | with compression and with caching of static resources enabled (N=20)
------------- | ------------- | -------------
Mean | 67.6  | 88.2
Standard Deviation  | 7.34  | 4.08

I used [this site](https://www.socscistatistics.com/tests/studentttest/default2.aspx) to test for statistical significance. In a two-tailed T-test, the t-value is -11.03609. The p-value is < .00001. The result is significant at p < .01.

### Conclusion  ###

There appears to be worthwhile performance improvement on FlightAware when enabling compression and caching of static resources on lighttpd. Recommend that the project maintainers consider enabling compression and caching of static resources on lighttpd.

mutability, 2022-02-27:
I appreciate that you've spent a decent amount of effort on this, but I think this is a solution in search of a problem.

Even without a cache-control header I'd expect the browser to be caching most/all of the static content. Full loads of the static content will be infrequent - skyaware gets loaded up then left to run, and the content is probably already cached. Initial load times (excluding history load) are under a second already. If you're worried about bandwidth / load times / etc then the _dynamic_ content (and the initial history load) is orders of magnitude larger. What's the _absolute_ decrease in warm-cache load time?

As I said earlier I'd really need to see a concrete proposal. That means a PR.

resuna, 2022-03-02:
I agree with mutability - these are minor improvements, and even if they were big improvements (multiple orders of magnitude) the performance of the RPI, particularly older models, means that in practice it's not network throughput that's the bottleneck. The main bottleneck is latency due to computational load on the pi itself.

But I don't know how these benchmarks account for latency and throughput. I don't know what the score in this benchmark actually means. My personal experience is that response time on maps on the piaware is already good on the local network, and that benchmark is more compelling for me than an undefined score.

cqueern, 2022-08-23:
Hello all. Any objections if I close this Issue as Won't Fix?

cqueern, 2022-08-25:
Closing as Won't Fix

closed 2022-08-25 by cqueern