flightaware/piaware #83: skyaware978 404

EasternPA, 2022-03-01:
Was the skyaware978 view removed? I can still get to the SkyAware map from the status page but SkyAware978 is now 404.

eric1tran, 2022-03-01:
Not in our officially released versions. What piaware/dump978-fa version are you on and what is the URL you're visiting?

EasternPA, 2022-03-01:
7.1

http://<hostip>:<skyaware port>/skyaware/ - works fine as always
http://<hostip>:<skyaware port>/skyaware978/ - is now 404 for me

if it is still working for everyone else, then something broke at my docker image provider and I'll keep digging.

mutability, 2022-03-01:
The package provides a lighttpd config fragment for that URL: https://github.com/flightaware/dump978/blob/33502c268cfa5efe18d2580bde85618fcdfefdd2/debian/lighttpd/89-skyaware978.conf#L7

and that config hasn't changed in a couple of versions

EasternPA, 2022-03-01:
Thank you for adding the tip. Yes, I expose port 80 as 8080. I looked at the code you linked to above and in the container to see if it got built. The container doesn't have the `/usr/share/skyaware978` folder that the alias statement in the lighttpd.conf snippet you linked to is expecting, only `/usr/share/skyaware`. That explains the 404 only for the `/skyaware978/` alias while `skyaware` still works fine. I do have `/usr/share/dump1090-fa` and `/usr/share/dump978-fa` folders in there with `html` folders located within. Just no `skyaware978` folder.

I'll have to see why the skyaware978 snippet is not making it into the container build. Thanks for your help.

closed 2022-03-01 by EasternPA

mutability, 2022-03-01:
`/usr/share/skyaware978` is part of the slyaware978 package: https://github.com/flightaware/dump978/blob/33502c268cfa5efe18d2580bde85618fcdfefdd2/debian/skyaware978.install#L1

EasternPA, 2022-03-01:
I copied /usr/share/dump978-fa into /usr/share/skyaware978 and that fixed it. Thanks again for your tips.