flightaware/piaware #86: piaware doesn't appear to reconnect after adept server side closes connection?

conorpwr, 2022-08-04:
I will admit up front that I'm using the https://github.com/sdr-enthusiasts/docker-piaware version of piaware and this may be out of your control. However I noticed that I had an unhealthy container and wasn't pushing data to flightaware any more. Looking at the logs revealed this:

```
2022-08-03 18:08:16.950969071  [piaware] 2022/08/03 18:08:16 1376580 msgs recv'd from dump1090 (1908 in last 5m); 1375968 msgs sent to FlightAware
2022-08-03 18:08:16.951401949  [piaware] 2022/08/03 18:08:16 10062 msgs recv'd from socat (5 in last 5m); 10048 msgs sent to FlightAware
2022-08-03 18:08:56.057050858  [piaware] 2022/08/03 18:08:56 Lost connection to adept server at 206.253.84.193/1200: server closed connection
```

I killed the piaware process within the container and let the supervisor restart it. Everything appears to come back as normal:

```
2022-08-03 23:10:12.025717204  [piaware] 2022/08/03 23:10:12 ****************************************************
2022-08-03 23:10:12.025722758  [piaware] 2022/08/03 23:10:12 piaware version 7.2 is running, process ID 40040
2022-08-03 23:10:12.027418624  [piaware] 2022/08/03 23:10:12 your system info is: Linux 4cf204e3e7e2 5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64 GNU/Linux
..... 
and so on
```

Shouldn't piaware attempt to reconnect at some point even if the error is server closed connection? I note there are other connection failures which appear to have retry logic in place.

```
2022-08-03 16:07:40.008802737  [piaware] 2022/08/03 16:07:40 Connecting to FlightAware adept server at piaware.flightaware.com/1200
2022-08-03 16:07:40.011843309  [piaware] 2022/08/03 16:07:40 Connection to adept server at piaware.flightaware.com/1200 failed: couldn't open socket: Temporary failure in name resolution
2022-08-03 16:07:40.012038455  [piaware] 2022/08/03 16:07:40 reconnecting in 5 seconds...
2022-08-03 16:07:45.012597608  [piaware] 2022/08/03 16:07:45 Connecting to FlightAware adept server at piaware.flightaware.com/1200
2022-08-03 16:07:45.015355554  [piaware] 2022/08/03 16:07:45 Connection to adept server at piaware.flightaware.com/1200 failed: couldn't open socket: Temporary failure in name resolution
2022-08-03 16:07:45.015543990  [piaware] 2022/08/03 16:07:45 reconnecting in 5 seconds...
2022-08-03 16:07:50.015957079  [piaware] 2022/08/03 16:07:50 Connecting to FlightAware adept server at 206.253.80.197/1200
2022-08-03 16:07:50.016448144  [piaware] 2022/08/03 16:07:50 Connection to adept server at 206.253.80.197/1200 failed: network is unreachable
2022-08-03 16:07:50.016702010  [piaware] 2022/08/03 16:07:50 reconnecting in 5 seconds...
2022-08-03 16:07:55.017258819  [piaware] 2022/08/03 16:07:55 Connecting to FlightAware adept server at 206.253.84.193/1200
2022-08-03 16:07:55.082263150  [piaware] 2022/08/03 16:07:55 Connection with adept server at 206.253.84.193/1200 established
2022-08-03 16:07:55.221978185  [piaware] 2022/08/03 16:07:55 TLS handshake with adept server at 206.253.84.193/1200 completed
2022-08-03 16:07:55.222615483  [piaware] 2022/08/03 16:07:55 FlightAware server certificate validated
2022-08-03 16:07:55.222624778  [piaware] 2022/08/03 16:07:55 encrypted session established with FlightAware
```

mutability, 2022-08-04:
This is probably due to an unresolved bug in third-party `tcl-tls` code which sometimes causes the entire piaware process to hang. Unfortunately there's not much we can do about that directly while the bug remains unfixed. I have some changes pending for 8.0 which will introduce a systemd watchdog to try to catch it, but if you're running under docker not via systemd then that may be ineffective.

conorpwr, 2022-08-06:
Ah well that is unfortunate. I'll see if there is anything worth doing in the container to auto restart the piaware process when it locks up.

closed 2023-07-26 by eric1tran