flightaware/piaware #89: Error loop on boot prevent access to piaware

sbsrouteur, 2022-09-10:
Hi, 

After a power outage, yesterday, my piaware did not restart.
Connected a video cable to console to find that an error in a /var/lib/dpkg/status crashes a script and the boot sequence does not seem to go far enough for ssh access to console login.

It seems to just loop on this. the device nic lights up, but does not seem to go far enough to request a dhcp address from the lan.

See attached screenshot for more detailed info on the error
[image]

Is there a way to get out of this loop?

Thank you for your help.

mutability, 2022-09-10:
This is filesystem / sdcard damage. I'd suggest recovering from a backup or
reimaging, perhaps with a new sdcard.

sbsrouteur, 2022-09-10:
sd was damaged.  Not a bug on piaware side.

closed 2022-09-10 by sbsrouteur