flightaware/piaware #92: MLAT problems with version 8.0~dev (and Python 3.9?)

by, 2022-11-06:
The current build seems to not work with my bullseye environment with Python 3.9 on my Raspberry Pi, fa-mlat-client exits with EXIT 255 – see below:

Nov  6 17:15:22 server piaware[947]: Starting multilateration client: /usr/lib/piaware/helpers/fa-mlat-client --input-connect localhost:30005 --input-type dump1090 --results beast,connect,localhost:30104 --results beast,listen,30105 --results ext_basestation,listen,30106 --udp-transport 2620:13d:c004:11::198:16950:2731017259
Nov  6 17:15:22 server piaware[947]: mlat-client(162519): Traceback (most recent call last):
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):   File "/home/pi/piaware_builder/package-bullseye/debian/venv/lib/python3.9/site-packages/cx_Freeze-6.8.3-py3.9-linux-aarch64.egg/cx_Freeze/initscripts/__startup__.py", line 113, in run
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):     module_init.run(name + "__main__")
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):   File "/home/pi/piaware_builder/package-bullseye/debian/venv/lib/python3.9/site-packages/cx_Freeze-6.8.3-py3.9-linux-aarch64.egg/cx_Freeze/initscripts/Console.py", line 15, in run
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):     exec(code, module_main.__dict__)
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):   File "/home/pi/piaware_builder/package-bullseye/debian/venv/bin/fa-mlat-client", line 4, in <module>
Nov  6 17:15:22 server piaware[947]: mlat-client(162519):     __import__('pkg_resources').run_script('MlatClient==0.2.12', 'fa-mlat-client')
Nov  6 17:15:22 server piaware[947]: mlat-client(162519): ModuleNotFoundError: No module named 'pkg_resources'
Nov  6 17:15:22 server piaware[947]: got EOF from multilateration client
Nov  6 17:15:22 server piaware[947]: fa-mlat-client exited with EXIT 255

eric1tran, 2022-11-21:
Can you provide all the steps to reproduce this? Did you build this on a standard Bullseye image?

by, 2022-11-21:
Yes, plain vanilla setup of bullseye; built and installed according to the instructions ([piaware_builder](https://github.com/flightaware/piaware_builder) plus https://github.com/flightaware/dump1090/blob/dev/README.md).

by, 2022-11-21:
After installation and reboot, I find above messages in the log

by, 2022-11-21:
Attached the built arm64 package [piaware_8.0~dev_arm64.deb.zip](https://github.com/flightaware/piaware/files/10061117/piaware_8.0.dev_arm64.deb.zip)

eric1tran, 2022-11-22:
> ModuleNotFoundError: No module named 'pkg_resources' 

Try installing python3-setuptools if you don't have it installed.

I tried your the package you built and it's complaining about some missing dependencies. I built it manually myself and it seems to be working ok. 

Try [](url)
[piaware_8.0~dev_arm64.deb.zip](https://github.com/flightaware/piaware/files/10068903/piaware_8.0.dev_arm64.deb.zip)

by, 2022-11-22:
Thanks and it was already installed: python3-setuptools is already the newest version (52.0.0-4)

closed 2022-12-28 by by

rromanchuk, 2023-01-08:
@by you ever get up and running on arm64? looks like I'm in tcl dependency hell

```
sudo dpkg -i piaware_8.2_arm64.deb 
Selecting previously unselected package piaware.
(Reading database ... 68487 files and directories currently installed.)
Preparing to unpack piaware_8.2_arm64.deb ...
Unpacking piaware (8.2) ...
dpkg: dependency problems prevent configuration of piaware:
 piaware depends on tcl-tls (>= 1.7.22-2+fa1); however:
  Version of tcl-tls:arm64 on system is 1.7.22-2.

dpkg: error processing package piaware (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
Processing triggers for rsyslog (8.2102.0-2+deb11u1) ...
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
```

by, 2023-01-08:
Yes, @rromanchuk , both with @abcd567a's packaged version from https://github.com/abcd567a/rpi and also compiling it myself via the instructions at https://github.com/flightaware/piaware_builder on https://github.com/flightaware/piaware/tree/dev and https://github.com/flightaware/dump1090/tree/dev.
`tcl-wise`, I have those packages installed:
- `tcl-dev/stable,now 8.6.11+1 arm64 [installed]`
- `tcl-tls/unknown,now 1.7.22-2+fa1 arm64 [installed]`
- `tcl8.6-dev/stable,now 8.6.11+dfsg-1 arm64 [installed]`
- `tcl8.6/stable,now 8.6.11+dfsg-1 arm64 [installed]`
- `tcl/stable,now 8.6.11+1 arm64 [installed,automatic]`
- `tcllib/stable,stable,now 1.20+dfsg-1 all [installed]`
- `tclx8.4/stable,now 8.4.1-4 arm64 [installed]`

Don't forget the comment
> If you are running a version of Debian that includes packages newer than those in Debian Buster, then the version constraint won't matter - but you should still check that you have a package version that actually fixes the problem!

P.S.: Starting point was Raspberry Pi OS light from https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit
P.S.2: The only remaining question is why MLAT does not run with my own build but @abcd567a's...

abcd567a, 2023-01-08:
My pre-built package for 64-bit bullseye for RaspberryPi OS (https://github.com/abcd567a/rpi) works without any issue of MLAT because before building piaware package, I did NOT install tcl-tls from repository by command "apt install tcl-tls". Instead, I built it using flightaware source code "tcltls-rebuilt", installed tcl-tls package built, and then only proceeded with building package for piaware using flightaware source coude piaware_builder.

### Building & Installing tcl-tls from source code. 
**(1) Installing tcl-tls dependencies**
```
sudo apt install -y libssl-dev
sudo apt install -y tcl-dev
sudo apt install -y chrpath
```

**(2)Cloning tcltls-rebuild source code and building the package**
```
git clone https://github.com/flightaware/tcltls-rebuild
cd tcltls-rebuild

./prepare-build.sh bullseye
cd package-bullseye
sudo dpkg-buildpackage -b --no-sign
```

**(3) Installing tcl-tls package**
`````

cd ../
sudo dpkg -i tcl-tls_*.deb
```

rromanchuk, 2023-01-08:
thanks this is probably where things went wrong. I ran adsbexchange setup scripts first and noticed broken mlat, and also apt installed tcl-tls before running piaware_builder.  

I missed your images when digging around, going to just start over and use them

rromanchuk, 2023-01-08:
@abcd567a thanks for doing this, up and running in literally 2 minutes from flash

by, 2023-01-08:
Thanks @abcd567a ; running bullseye, I've got confused by the part "If you are running a version of Debian that includes packages newer than those in Debian Buster, then the version constraint won't matter" of
> If you are running a version of Debian that includes packages newer than those in Debian Buster, then the version constraint won't matter - but you should still check that you have a package version that actually fixes the problem!

by, 2023-01-08:
P.S.: Actually, not so much a change: while MLAT seems to work here in general, too, the original issue, it being marked as red at https://de.flightaware.com/adsb/stats/user/[my username] still persists...