flightaware/piaware #97: AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'

idatum, 2023-10-21:
I periodically see an exception on my Piaware instance. Looks like a module version mismatch? Here are my logs:

[piaware] 2023/10/21 19:06:07 mlat-client(174): outlier detected with ts: 6269.091, last_ts: 6269.063, ts_elapsed: 0.027, sys_elapsed: 1.503 (values in seconds)
[piaware] 2023/10/21 19:06:07 mlat-client(174): Unexpected exception on connection to localhost:30005
[piaware] 2023/10/21 19:06:07 mlat-client(174): Traceback (most recent call last):
[piaware] 2023/10/21 19:06:07 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 88, in read
[piaware] 2023/10/21 19:06:07 mlat-client(174):     obj.handle_read_event()
[piaware] 2023/10/21 19:06:07 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 425, in handle_read_event
[piaware] 2023/10/21 19:06:07 mlat-client(174):     self.handle_read()
[piaware] 2023/10/21 19:06:07 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/receiver.py", line 214, in handle_read
[piaware] 2023/10/21 19:06:07 mlat-client(174):     self.coordinator.input_received_messages(messages)
[piaware] 2023/10/21 19:06:07 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 328, in input_received_messages
[piaware] 2023/10/21 19:06:07 mlat-client(174):     handler(message, now)
[piaware] 2023/10/21 19:06:07 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 350, in received_timestamp_jump_event
[piaware] 2023/10/21 19:06:07 mlat-client(174):     self.server.send_clock_jump()
[piaware] 2023/10/21 19:06:07 mlat-client(174):     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[piaware] 2023/10/21 19:06:07 mlat-client(174): AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'
[piaware] 2023/10/21 19:06:07 mlat-client(174): Lost connection to localhost:30005

mutability, 2023-10-21:
You seem to be running a 3rd party fork of mlat-client with piaware.
Usually piaware will include its own separate copy of mlat-client to avoid
this sort of breakage. How did you build/install your piaware?

idatum, 2023-10-21:
I'm running a container of image from sdr-enthusiasts/docker-piaware repo. I figured that might be the case regarding breakage, thanks.

I'll open an issue with sdr-enthusiasts/docker-piaware.

closed 2023-10-21 by idatum

mutability, 2023-10-22:
Yeah, this is because that docker container is using the wrong mlat-client.
I'll comment over there.