flightaware/piaware #100: statusfile.tcl is not setting type = flightfeeder

chuckwolber, 2024-02-11:
[This commit](https://github.com/flightaware/dump1090/commit/a0b0038df7ffe5d7021d6f3e73e6a436a58740e9) made to  [dump1090-fa](https://github.com/flightaware/dump1090) turned on a check in the `dump1090-fa` web interface to look for a `type` variable in `status.json`. When `type = "flightfeeder"`, a "FlightFeeder SkyAware" logo should appear on the web page and the page title will change to "FlightFeeder SkyAware". Otherwise a "PiAware SkyAware" logo appears and the page title is set to "PiAware SkyAware".

Since `piaware` is acting in the role of flight feeder by consuming data from `dump1090-fa` and pushing it to FlightAware, is it reasonable to expect that the `status.json` produced by `piaware` is indeed the same `status.json` being consumed by `dump1090-fa`?

If so, it does not appear as if `piaware` is actually adding the `type` variable to the `status.json`, nor does it appear as if `dump1090-fa` is able to properly access the `status.json` produced by `piaware`.

The following three patches will resolve the issue and get the correct logo and page title to appear. The first patch applies to the latest version of the `piaware` project. The latter two patches are intended for the latest version of `dump1090-fa`. I can submit a separate issue to the `dump1090-fa` project if desired.

```
diff --git a/programs/piaware/statusfile.tcl b/programs/piaware/statusfile.tcl
index ed7241c..8bb7002 100644
--- a/programs/piaware/statusfile.tcl
+++ b/programs/piaware/statusfile.tcl
@@ -69,7 +69,9 @@ proc build_status {} {
        # site UUID, if unclaimed
        if {[info exists ::feederID] && [info exists ::loggedInUser] && $::loggedInUser eq "guest"} {
                set data(unclaimed_feeder_id) [::json::write string $::feederID]
-       }
+       } else {
+        set data(type) [::json::write string "flightfeeder"]
+    }
 
        # piaware: our own health
        set data(piaware) [status_entry "green" "PiAware $::piawareVersionFull is running"]
```

```
diff --git a/public_html/script.js b/public_html/script.js
index 7217613..2ed02ed 100644
--- a/public_html/script.js
+++ b/public_html/script.js
@@ -2527,7 +2527,7 @@ function toggleLayer(element, layer) {
 
 // check status.json if it has a serial number for a flightfeeder
 function flightFeederCheck() {
-    $.ajax('/status.json', {
+    $.ajax('status.json', {
         success: function(data) {
             if (data.type === "flightfeeder") {
                 isFlightFeeder = true;
@@ -2538,7 +2538,7 @@ function flightFeederCheck() {
 }
 
 function setStatsLink() {
-        $.ajax('/status.json', {
+        $.ajax('status.json', {
                 success: function(data) {
                     if (data.unclaimed_feeder_id) {
                         var claim_link = "https://flightaware.com/adsb/piaware/claim/" + data.unclaimed_feeder_id;
```

```
diff --git a/debian/lighttpd/89-skyaware.conf b/debian/lighttpd/89-skyaware.conf
index 56f44d1..3d108f8 100644
--- a/debian/lighttpd/89-skyaware.conf
+++ b/debian/lighttpd/89-skyaware.conf
@@ -21,6 +21,7 @@
 # server.modules += ( "mod_alias" )
 
 alias.url += (
+  "/skyaware/status.json" => "/run/piaware/status.json",
   "/skyaware/data/" => "/run/dump1090-fa/",
   "/skyaware/data-978/" => "/run/skyaware978/",
   "/skyaware/" => "/usr/share/skyaware/html/"
@@ -34,6 +35,7 @@ url.redirect += (
 # Listen on port 8080 and serve the map there, too.
 $SERVER["socket"] == ":8080" {
   alias.url += (
+    "/status.json" => "/run/piaware/status.json",
     "/data/" => "/run/dump1090-fa/",
     "/data-978/" => "/run/skyaware978/",
     "/" => "/usr/share/skyaware/html/"
```

mutability, 2024-02-12:
FlightFeeders are specific hardware manufactured by FlightAware (https://www.flightaware.com/adsb/flightfeeder/). They run a software image that includes dump1090-fa, piaware, and other additional FlightFeeder-specific software. Amongst other things, the additional FlightFeeder-specific software provides a `/status.json` that provides overall status information for the whole system, separate to the piaware `status.json`

This change to dump1090-fa was so that skyaware is aware of when it's running on FlightFeeder hardware, and will display "FlightFeeder SkyAware" versus self-installed piaware builds which will show "PiAware SkyAware".

so it's expected that

a) piaware itself will not include a `type` field in the piaware `status.json`;
b) dump1090-fa's skyaware lighttpd doesn't export a `status.json`;
c) the skyaware display on a self-installed dump1090-fa / piaware install will show "PiAware SkyAware"

So I think this is all working as expected, unless there's some specific problem you ran into?

chuckwolber, 2024-02-13:
Copy that. Thank you for the clear explanation. I took a close look at the code and the `status.json` stood out to me as a curious loose-end with no obvious explanation. My natural conclusion was that `status.json` was a way for `piaware` to communicate status back to the `dump1090-fa` web interface. Based on your explanation, this was a bad assumption and the likeness is only coincidental.

Is it worth somehow noting that in a code comment or project documentation?