flightaware/piaware #101: buffer overflow

davxiao, 2024-04-13:
Latest version running on pi Zero W, got following error from log

```
Apr 12 22:18:20 pi-star piaware[1604]: creating pidfile /run/piaware/piaware.pid
Apr 12 22:18:20 pi-star piaware[1604]: ****************************************************
Apr 12 22:18:20 pi-star piaware[1604]: piaware version 9.0.1 is running, process ID 1604
Apr 12 22:18:21 pi-star piaware[1604]: your system info is: Linux pi-star 5.10.17+ #1414 Fri Apr 30 13:16:27 BST 2021 armv6l GNU/Linux
Apr 12 22:18:21 pi-star piaware[1604]: *** buffer overflow detected ***: /usr/bin/piaware terminated
```

mutability, 2024-04-13:
possibly #51 again (tcl bug when IPv6 is disabled)

check that IPv6 is enabled?

davxiao, 2024-04-13:
Problem solved. It seems to be ipv6 related. 
Would like to report back that I have tested 2 approaches below and both worked.

- Enable ipv6. `sudo vi /etc/sysctl.conf` and make sure the line is there `net.ipv6.conf.all.disable_ipv6 = 0` and reboot.
or 
- in `/etc/hosts` remove any line that is ipv6