flightaware/piaware #103: Errors for PiAware in syslog

hornjt75862, 2024-08-21:
I am getting the following errors repeatedly in my syslog file and was wondering if I could some assistants with how to solve it:
`Aug 21 14:26:37 PiAware rbfeeder[3139]: rtlsdr: error opening the RTLSDR device: Device or resource busy
Aug 21 14:26:39 PiAware rbfeeder[516]: #033[35m[#033[33m2024-08-21 14:26:39#033[35m]#033[0m  Can't connect to 978 source (127.0.0.1:28380). Waiting 5 second...
Aug 21 14:26:40 PiAware rbfeeder[3145]: Wed Aug 21 14:26:40 2024 CDT  dump1090-rb 1.0.10 starting up.
Aug 21 14:26:40 PiAware rbfeeder[3145]: rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001000)
Aug 21 14:26:40 PiAware rbfeeder[3145]: usb_claim_interface error -6
`
I was wondering if I could get some suggestions as to how to fix the issue.

HNJAMeindersma, 2024-08-26:
Is you PiAware not working properly?
It seems the RTLSDR device is marked busy (aka in use). Do you have any other services running that uses the RTLSDR dongle?

hornjt75862, 2024-08-26:
It looks like I was able to get it working by reinstalling  feeder
software. Thank you