flightaware/piaware #104: All about Bookworm

rmyadsk, 2024-09-21:
Hi all,

I'd like to start here a consolidated discussion about piaware's compatibility with Raspian Bookworm, which was released in October 2023. I've been cruising on Bullseye for some time, and "apt-get full-upgrade" results in a bunch of zeroes.

I'm currently deployed over on the right-hand side of the Atlantic (one receiver, in my residence here) though I manage two receivers back in the US at close friends' in two locations in the mid-Atlantic area. It so happens I'll be taking some time off to reconnect with them all during this holiday season - maybe my last opportunity to do so for a year or or more - so I'm interested in getting a long-term roadmap about where this project is (or is not) going with Bookworm.

My hardware are Pi Zeroes running an oleo of "PiAware (SD Card) 9.0.1" and "PiAware (Debian Package Add-on) 9.0.1". What's the best long-term solution here?

wcbonner, 2024-09-21:
I had been running an install of piaware on a PiZeroW running raspian Bookworm and updating it regularly with the command:
```
sudo apt-get update && sudo apt-get upgrade -y && sudo apt-get dist-upgrade -y && sudo apt autoremove -y
```

Because I wanted to standardize all my systems to running 64bit I recently swapped out the hardware with a Zero2. I still use the same update sequence periodically. I now have several PiZero devices sitting idle and unused.  

```
wim@WimPiZero2W-Brett:~ $ cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi Zero 2 W Rev 1.0
wim@WimPiZero2W-Brett:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

wcbonner, 2024-09-21:
I installed PiAware on my Raspberry Pi OS Lite (64 Bit) image that I'd built using the Raspberry Pi Imager. I used the following command at the ssh prompt: 
```
wget https://www.flightaware.com/adsb/piaware/files/packages/pool/piaware/f/flightaware-apt-repository/flightaware-apt-repository_1.2_all.deb && sudo dpkg -i flightaware-apt-repository_1.2_all.deb
```
and then copied my old configuration from the previous SD card.

eric1tran, 2024-09-21:
@rmyadsk Your option for Bookworm right now is using the standard Raspberry Pi OS Bookworm image and install the piaware debian package using the method @wcbonner mentioned.

We anticipate a Bookworm PiAware SD card image being released sometime this year.

rmyadsk, 2024-09-22:
Thanks @wcbonner and @eric1tran for the comprehensive info - the FlightAware website still points to Raspbian 8.2. This is just what I needed!

rmyadsk, 2024-12-10:
@eric1tran pinging for outlook on a Bookworm PiAware SD card image this year.