flightaware/piaware #107: piaware 10.1 wireless will fail if the password has an apostrophe in it

bp0, 2025-05-23:
Updated to 10.1 SD card image and copied my working config over, but it would not reconnect.
Assumed it has something to do with the change from ifupdown to NetworkManager in bookworm.
It seems the piaware scripts truncate at the `'` if it exists in the password, certainly a bug.
Fixed it in the NetworkManager connection settings, but after reboot, piaware changed it back to the broken version and it would still not connect.

sjemmett, 2025-05-25:
10.1 also fails when password is a passphrase that has multiple spaces and the entire passphrase is enclosed in either ' or ".

resuna, 2025-05-27:
This line in particular looks smoke-emitting:

https://github.com/flightaware/piaware-support/blob/master/flightaware_piaware_config/src/flightaware_piaware_config/piaware_config.py#L290

mutability, 2025-05-27:
Dodgy ad-hoc parsing is unfortunately the name of the game here.

The canonical parser is this tcl code: https://github.com/flightaware/piaware/blob/0cea9bbe483f8a15c1b6a5e4883072d8df1ff8b4/package/fa_piaware_config.tcl#L582-L629

which (to simplify) handles two cases:

* An option where the option value string starts with `'` or `"`. In this case, the option value is everything up to the next instance of the same quote character, except you can backslash-escape any character. Everything after the closing quote is entirely ignored.
* An option where the option value string *doesn't* start with a quote. In this case, the option value ends at end-of-line or a `#` comment character, and no backslash-escaping is done, and the option value has surrounding whitespace trimmed.

i.e.

```
wireless-password "abcd e f ''' \" some more "
```

should produce an option value of `abcd e f ''' " some more ` (with a trailing space)

```
wireless-password abcdef' ghi   # this is my password!
```
should produce an option value of `abcdef' ghi` (with no trailing whitespace)

From a quick look the python implementation is missing the subtlety that a _matching_ quote is needed to terminate the quoted case, _not_ either type of quote

mutability, 2025-05-27:
A possible workaround (that should continue to work after this bug is fixed) is to backslash-escape all quotes of either type within a quoted string.

TechnicalLee, 2025-06-08:
Also had this problem, but with a backslash in the password instead of an apostrophe. Backslashes must also escaped in the piaware-config file, even if not using quotes.

`pass\word` has to be changed to `pass\\word`

nikeflight, 2025-07-01:
Hi all, 

We've pushed a new image 10.2 that should fix these wifi issues.

Side note:
NetworkManager expects any `\` to be escaped; so we added logic to escape them automatically.
It seems like that's the only special character that needs any special treatment.

bp0, 2025-07-04:
The full value for `wireless-password` in my old (pre-10.1) config file was copied correctly into `/etc/NetworkManager/system-connections/wireless.nmconnection` after reboot into 10.2. 
So it seems the reported bug is fixed, for me at least. 
Thanks.