flightaware/piaware #109: Trixie

CdeMills, 2025-07-19:
Ring ... Ring ... Debian has just release trixie, and so did raspian (on July, 16th, 2025)

I tried the switch, only to discover that the update would wipe out piaware. 

Any hint on keeping or updating it in Trixie ?

TIA

Pascal

wcbonner, 2025-07-20:
@CdeMills As far as I can tell, Trixie is nearing release but not yet released. https://wiki.debian.org/DebianReleases

bartgrefte, 2025-11-29:
> > [@CdeMills](https://github.com/CdeMills) As far as I can tell, Trixie is nearing release but not yet released. https://www.debian.org/releases/
> 
> Raspberry released their version of Trixie on October 2nd, so hopefully the piaware installer will be updated to support trixie soon. https://www.raspberrypi.com/news/trixie-the-new-version-of-raspberry-pi-os/

After I did the upgrade from Bookworm to Trixie I got
```
Warning: http://flightaware.com/adsb/piaware/files/packages/dists/bookworm/InRelease: Policy will reject signature within a year, see --audit for details

code (1), error message is:
   Signing key on 4E49CAF6B9F2BF8FC240E4BAB931BB28DE85F0DD is not bound:
              No binding signature at time 2025-06-30T15:04:16Z
     because: Policy rejected non-revocation signature (PositiveCertification) requiring second pre-image resistance
     because: SHA1 is not considered secure since 2026-02-01T00:00:00Z

```

So I did
```
wget https://www.flightaware.com/adsb/piaware/files/packages/pool/piaware/f/flightaware-apt-repository/flightaware-apt-repository_1.2_all.deb
sudo dpkg -i flightaware-apt-repository_1.2_all.deb
```
( from https://www.flightaware.com/adsb/piaware/install) to have the sources file updated. Instead I got a 2nd problem:
`Error: The repository 'http://flightaware.com/adsb/piaware/files/packages trixie Release' does not have a Release file.`

So the installer creates a sources file with Trixie, but since there is no releases file, that doesn't work. Seems someone forgot something....

bartgrefte, 2025-12-04:
@mik-at .... then why is the flightaware-apt-repository deb adding
"http://flightaware.com/adsb/piaware/files/packages **trixie** Release" ?
Seems they are aware that Trixie has been released, but instead of supporting it they are creating an apt sources file that doesn't work.

closed 2026-04-08 by mutability

scottjpalmer, 2026-05-16:
> It'll be pushed out a week. The packages are ready, we're updating our apt repositories for Trixie support and will need to do some upgrade testing on it.

Sorry to pester but I'm keeen for an update please @eric1tran